FreeCAD/FreeCAD #12149: [Problem] opensbp processor doesn't recognize G54 code

kzembower, 2024-01-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Creating an opensbp file generates this code (in the terminal when FreeCAD is started by the command line):
```
post: opensbp(/home/kevin/CAD_Mechanical/GoldenRatioRectangle_Chalkboard_inch.sbp, --inches)
postprocessing...
I don't know what the hell the command: G54 means.  Maybe I should support it.
done postprocessing.
```
G54 sets the first Work Coordinate System (WCS). This is checked off by default in the Path:Job:Output window. Without any WCS checked off, no gcode is generated; the output file consists of just the headers, pre- and post-amble. However, if this error message is ignored,  the output file is generated normally and it seems to be run-able and correct (haven't run on ShopBot yet).

The Path:Job:Output window also contains check-off boxes for the WCS G54-G59.9. I suggest that these WCS either be processed correctly also, or a tool-tip or other notice appear that only G54 is supported by the opensbp processor.

### Full version info

```shell
[code]
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.62
  * FeedsAndSpeeds 0.5.0
  * Assembly3 0.12.1 (Disabled)
  * Help 1.0.3
  * Assembly4 0.50.6
  * woodworking 0.21.1.33694
[/code]
```

### Subproject(s) affected?

Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-03-13:
Hi @mlampert any chance you can take a look at some of these openbsp post-processor tickets that were recently opened ?

luzpaz, 2025-06-02:
@kzembower are you still active ? If we find a dev to fix this can you test it  ?

sliptonic, 2025-06-03:
Don't confuse the internal Path commands with gcode.  They are *mostly* the same thing but semantically a little different.  We use a subset of gcode commands to store the internal toolpath representation. This is generally very closely aligned with RS274 standard which is also what linuxcnc uses.   

The path commands are converted to *real* gcode at post-processing.  
The G54 sets the first (default) coordinate system to use.  So it's the job of a post-processor receiving a G54 to convert it into the specific gcode that the machine control needs to set the first WCS touch-off point.

I wrote the original shopbot post but haven't used one in many years and  I can't even remember how it handles coordinate systems. 😆