FreeCAD/FreeCAD #12152: [Problem] Navigation cube orientation cannot be changed

pierreporte, 2024-01-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

TOP is always oriented along Z+, FRONT is always oriented along Y−, etc. Depending on the project, user preferences, and company policy, this orientation may be incoherent with the design intent. For example, a vehicle is usually designed with the X axis along the direction of motion, and for planes the zero is at the front tip.

There should be a document setting to set where FRONT and TOP points to. It could be in the document information dialog like units.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-04-22:
@maxwxyz for the DWG project.

ElementW, 2026-09-15:
Note that to be consistent across the UI and code, not just the navicube needs to be changed, but also fundamentally all code referring to any of `front`, `rear`, `left` and `right`, to make those definitions dependent on the meaning given by the active document (or object thereof if we allow for that granularity).