FreeCAD/FreeCAD #12153: [Problem] Installing FreeCAD 21.1 removes KiCAD 7.0. on Ubuntu 22.04 - Reinstall of KiCAD is not possible.

schoenid, 2024-01-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Installing FreeCAD 21.1 removes KiCAD 7.0. on Ubuntu 22.04 - Reinstall of KiCAD is not possible.

```
$ sudo apt install freecad
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Die folgenden Pakete wurden automatisch installiert und werden nicht mehr benötigt:
  kicad-demos kicad-doc-en kicad-footprints kicad-libraries kicad-packages3d kicad-symbols kicad-templates libngspice-kicad libpcre2-32-0 libwxbase3.2-1
  libwxgtk-gl3.2-1 libwxgtk3.2-1 python3-wxgtk4.0 unixodbc
Verwenden Sie »sudo apt autoremove«, um sie zu entfernen.
Die folgenden zusätzlichen Pakete werden installiert:
  freecad-python3 libfreecad-python3-0.21 libocct-data-exchange-7.6 libocct-draw-7.6 libocct-foundation-7.6 libocct-modeling-algorithms-7.6
  libocct-modeling-data-7.6 libocct-ocaf-7.6 libocct-visualization-7.6
Vorgeschlagene Pakete:
  povray
Empfohlene Pakete:
  python3-opencamlib
Die folgenden Pakete werden ENTFERNT:
  kicad libocct-data-exchange-7.5 libocct-foundation-7.5 libocct-modeling-algorithms-7.5 libocct-modeling-data-7.5 libocct-ocaf-7.5
  libocct-visualization-7.5
Die folgenden NEUEN Pakete werden installiert:
  freecad freecad-python3 libfreecad-python3-0.21 libocct-data-exchange-7.6 libocct-draw-7.6 libocct-foundation-7.6 libocct-modeling-algorithms-7.6
  libocct-modeling-data-7.6 libocct-ocaf-7.6 libocct-visualization-7.6
0 aktualisiert, 10 neu installiert, 7 zu entfernen und 3 nicht aktualisiert.
Es müssen noch 0 B von 59.6 MB an Archiven heruntergeladen werden.
Nach dieser Operation werden 46.1 MB Plattenplatz freigegeben.
```
Trying to reinstall KiCAD:

```
$ sudo apt -f install --install-recommends kicad --fix-broken
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Einige Pakete konnten nicht installiert werden. Das kann bedeuten, dass
Sie eine unmögliche Situation angefordert haben oder, wenn Sie die
Unstable-Distribution verwenden, dass einige erforderliche Pakete noch
nicht erstellt wurden oder Incoming noch nicht verlassen haben.
Die folgenden Informationen helfen Ihnen vielleicht, die Situation zu lösen:

Die folgenden Pakete haben unerfüllte Abhängigkeiten:
 libocct-data-exchange-7.6 : Beschädigt: libocct-data-exchange-7.5 aber 1:7.5.2+dfsg1-0~202107020155~ubuntu22.04.1 soll installiert werden
 libocct-foundation-7.6 : Beschädigt: libocct-foundation-7.5 aber 1:7.5.2+dfsg1-0~202107020155~ubuntu22.04.1 soll installiert werden
 libocct-modeling-algorithms-7.6 : Beschädigt: libocct-modeling-algorithms-7.5 aber 1:7.5.2+dfsg1-0~202107020155~ubuntu22.04.1 soll installiert werden
 libocct-modeling-data-7.6 : Beschädigt: libocct-modeling-data-7.5 aber 1:7.5.2+dfsg1-0~202107020155~ubuntu22.04.1 soll installiert werden
 libocct-ocaf-7.6 : Beschädigt: libocct-ocaf-7.5 aber 1:7.5.2+dfsg1-0~202107020155~ubuntu22.04.1 soll installiert werden
 libocct-visualization-7.6 : Beschädigt: libocct-visualization-7.5 aber 1:7.5.2+dfsg1-0~202107020155~ubuntu22.04.1 soll installiert werden
E: Fehler: Unterbrechungen durch pkgProblemResolver::Resolve hervorgerufen; dies könnte durch zurückgehaltene Pakete verursacht worden sein.
```

So the de-installation of another software is a violation of a users privacy and may not be tolerable!
The FreeCAD installation has to be modified to respect this.

### Full version info

```shell
[code]
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Switzerland (de_CH)
Installed mods: 
  * MakerWorkbench 1.0.1
  * parts_library
  * Curves 0.6.8
  * freecad.gears 1.0.0
  * fasteners 0.4.56
  * frame 0.1.0
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mfraser, 2024-01-27:
If you want a quick fix.

```
apt download kicad
mkdir temp
dpkg-deb -R kicad_7.0.10~ubuntu22.04.1_amd64.deb temp/
```
Inside temp/DEBIAN, replace the control file with this one:
[control](https://github.com/FreeCAD/FreeCAD/files/14072759/control.txt)
```

dpkg-deb -b temp/ kicad_7.0.10~ubuntu22.04.2_amd64.deb
sudo apt install ./kicad_7.0.10~ubuntu22.04.2_amd64.deb
```

adrianinsaval, 2024-01-27:
this is expected, freecad ppa uses a more updated version of occt than available on the ubuntu repos. I am not willing to downgrade the occt version because the older version has known bugs and I don't want our issue tracker flooded with reports of already fixed bugs. What we could do is reach out to the kicad team and add occt 7.6.3 in their ppa too https://launchpad.net/~kicad/+archive/ubuntu/kicad-7.0-releases
@kkremitzki I see you made occt 7.5.2 packages for them at some point, how can we go about having that updated?

mfraser, 2024-01-27:
> this is expected, freecad ppa uses a more updated version of occt than available on the ubuntu repos. I am not willing to downgrade the occt version because the older version has known bugs and I [...]

According to Kicad's Mastodon account, they are planning on updating the nightly build to use 7.6 this week.

adrianinsaval, 2024-01-27:
Awesome, just jumped in and gave them links to our stuff if they want to use it

Fubbel, 2024-01-28:
You can select with synatics the earlyer Version .20 from Freecat, but someone removed some files for Ubuntu 22 and this earlier Version, so the dependency is not cleared...
Please restore the removed files...

adrianinsaval, 2024-03-20:
they seem to be using occt 7.6.3 too now so it should be coinstallable now

closed 2024-03-20 by adrianinsaval

emogenet, 2025-01-22:
This issue is still here on ubuntu 24.04 ...
Freecad 0.21.2 rev 33771 conflicts with kicad 8.0 because of some occt related things

Reproduce with these steps:

Get a clean ubuntu 24.04

add-apt-repository --yes ppa:kicad/kicad-8.0-releases

add-apt-repository ppa:freecad-maintainers/freecad-stable

apt install freecad

apt install -install-recommends kicad

watch the latter kick the former out of the machine.

mikhailmihalkov, 2025-02-03:
History repeats itself with KickDown 8 and Ubuntu PPA 
https://forum.kicad.info/t/installation-on-ubuntu-22-04/55363