FreeCAD/FreeCAD #12154: [Problem] Path: Clicking on "Check path for common errors" icon results in FreeCAD hanging

kzembower, 2024-01-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Clicking on the Path icon 'Check path for common errors (P, S)' causes FreeCAD to hang for multiple minutes (only ended when FreeCAD was killed). Choosing the same command from the Path menu causes the same symptom, with an addtional error message: `Reading failed from embedded file: PartShape7.bin (0 bytes, 2 bytes compressed)`

While FreeCAD is hung, top shows 100% CPU usage by freecad.

### Full version info

```shell
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.62
  * FeedsAndSpeeds 0.5.0
  * Assembly3 0.12.1 (Disabled)
  * Help 1.0.3
  * Assembly4 0.50.6
  * woodworking 0.21.1.33694
```

### Subproject(s) affected?

Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-01-28:
I don't have such hanging. Can you provide an example file?

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
```

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35733 (Git)
Build type: Debug
Branch: main
Hash: f75db0dbb2f48f3b486afa80afe9d75ab19970c9
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

----

kzembower, are you using 0.21.1 for a reason? Have you tried 0.21.2 [^1] or a weekly build? [^2]

[^2]: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
[^1]: https://github.com/FreeCAD/FreeCAD/releases/

kzembower, 2024-01-28:
Yes, _using the attached file_, I switch to the Path WB, click on the Job in the Model view, and click on the Check Path icon. It hangs, with top showing 100% CPU usage by freecad.

**Just learned that it's not possible to upload a *.FCStd file to the FreeCAD github site. "We don’t support that file type." Interesting. Please advise.**

I use 0.21.1 because it works for me, and I'm interested in producing parts, and, at this time, not necessicarily interested in being a developer.

Thanks for your help in looking into this.

-Kevin

kzembower, 2024-01-28:
Here's a Dropbox link where you can download my file: https://www.dropbox.com/scl/fi/ehmykamei8mp65h4jahn3/GoldenRatioRectangle_Chalkboard.FCStd?rlkey=ims394c6yptbwbbk0ze0f0orw&dl=0

Here's a copy of top with FreeCAD hung:
```
kevin@kevin-desktop:~/CAD_Mechanical$ top

top - 08:33:01 up 1 day, 18:29,  1 user,  load average: 1.27, 1.20, 0.69
Tasks: 337 total,   2 running, 334 sleeping,   0 stopped,   1 zombie
%Cpu(s): 21.1 us,  4.6 sy,  0.0 ni, 74.2 id,  0.1 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :  32016.1 total,  20114.4 free,   4611.6 used,   7290.1 buff/cache
MiB Swap:   2048.0 total,   2048.0 free,      0.0 used.  26849.2 avail Mem 

    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND  
  39042 kevin     20   0 3177948 448264 263988 R 100.0   1.4   9:46.59 freecad  
   1425 kevin     20   0 3907136 300648 123864 S   0.7   0.9  24:52.32 gnome-s+ 
   4841 kevin     20   0  694548  54844  41864 S   0.7   0.2   0:12.03 gnome-t+ 
```
Let me know if I can do anything else. Thanks, again, for looking at this.

-Kevin

xtemp09, 2024-01-28:
This version hangs too.

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35733 (Git)
Build type: Debug
Branch: main
Hash: f75db0dbb2f48f3b486afa80afe9d75ab19970c9
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

----

This problem comes from Python.

sliptonic, 2024-05-07:
Should be fixed with refactored Sanity.
https://github.com/FreeCAD/FreeCAD/pull/13667

closed 2024-05-07 by sliptonic