FreeCAD/FreeCAD #12156: [Problem] Tracing external geometry arc results in over constrained sketch.

ksdsksd, 2024-01-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I cannot find this exact issue.

Attempting draw the same arc(center and endpoints),  as the external geometry results in an over constrained sketch.
[image]
[image]
[image]

### Full version info

```shell
[code]
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60g
  * fcgear 1.0.0
  * Assembly4 0.12.5
  * Curves 0.5.4
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-01-29:
The 2nd coincident constraint fixes the radius a 2nd time. Try replacing it with a horizontal or vertical constraint.

ksdsksd, 2024-01-30:
The same steps in version 0.20, it becomes partially redundant and shows in green.
[image]

Roy-043, 2024-02-10:
Duplicate of #11852.

closed 2024-02-10 by Roy-043