FreeCAD/FreeCAD #12162: [Problem] Navigation: Rotation center moves while rotating

maxwxyz, 2024-01-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Despite fixing https://github.com/FreeCAD/FreeCAD/issues/11243 with https://github.com/FreeCAD/FreeCAD/pull/11279 the rotation center moves while orbiting the camera.
@Rexbas noticed this behavior not only in CAD navigation style around window center but in all navigation styles.

It seems to happen when you zoom in for details or navigate around larger scenes. The new rotation indicator makes it possible to see easily.
Here is a demo file. The rotational center is rotating and moving when you just try to rotate it:
[CAD-around-center-demo.zip](https://github.com/FreeCAD/FreeCAD/files/14075837/CAD-around-center-demo.zip)

Maybe this is a float issue as pointed out here: https://github.com/FreeCAD/FreeCAD/pull/11279#issuecomment-1913560755

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35449 (Git)
Build type: Release
Branch: main
Hash: f3fa2ba9c9b1999e5c32d3ae544481b6e17553bb
Python 3.11.7, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.1
  * CurvedShapes 1.0.5
  * Curves 0.6.22
  * Defeaturing 1.2.2
  * fasteners 0.5.0
  * FEMbyGEN 2.1.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.4.0
  * woodworking 0.21.2.33771
```

### Subproject(s) affected?

Core

### Anything else?

https://github.com/FreeCAD/FreeCAD/pull/11279

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-04-09:
@Rexbas it's very obvious when using the Optics WB as the default ray length is 1 mil. mm:

https://github.com/FreeCAD/FreeCAD/assets/6246609/7ef13ce1-f571-4d7d-80c1-4489d5fdd960

maxwxyz, 2024-04-11:
@Rexbas any idea what's the timeline for the coin release?

luzpaz, 2024-08-29:
What release version were we waiting for ?

Rexbas, 2024-08-30:
Both of my PRs in coin have been merged so whatever the next version is, probably 4.0.3.

luzpaz, 2024-08-30:
Made a request for an upstream release https://github.com/coin3d/coin/pull/519#issuecomment-2321858309

luzpaz, 2024-08-30:
@Rexbas will this issue be complete once upstream make a release with your changes ? (or is there something left here?)

Rexbas, 2024-08-31:
I don't think this issue can be fixed completely as it seems to be a floating point precision issue which occurs in very small or very big scenes. However, the specific case in the Optics WB is resolved with the fixes in coin and after reverting some temporary commits in FreeCAD. When there is a new coin release I will make a PR to revert these temporary commits.

maxwxyz, 2024-09-07:
@Rexbas niw with the new coin release what needs to be done? Are there other issues I can close now?

Rexbas, 2024-09-07:
@maxwxyz, see the linked PR.

cpresser, 2025-05-26:
I have a similar problem in 1.0.1. So not sure if I should re-open this issue or create a new one.
It is quite likely that this is a problem with my file. But I sure don't know which one. I did remove all features, just one feature and a few sketches remain in the file.

Here is the offending file:  [temp.FCStd.gz](https://github.com/user-attachments/files/20445137/temp.FCStd.gz)

I tried to record a video: 

https://github.com/user-attachments/assets/18cd9278-f984-4345-b8af-0ac247fd0af4

```
OS: KDE Flatpak runtime (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 878f0b8)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.12.10, Qt 6.9.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * kicadStepUpMod
  * kicadStepUpMod.backup1739127269.633388 (Disabled)

```

closed 2025-06-06 by kadet1090