FreeCAD/FreeCAD #12163: [Problem] SIGSEGV on creation of FEM Analysis Container

tarihatari, 2024-01-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create New Document
2. Switch to FEM Workbench
3. Click on `Analysis container'`
4. crash

Happens to me with all FreeCAD versions > 0.20.2 

### Full version info

```shell
OS: Ubuntu 22.04.3 LTS (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35744 (Git)
Build type: Release
Branch: main
Hash: 3d6fb38de1e856b9a975b6c8a635a0eb9131ac8e
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * lattice2 1.0.0
  * dodo 1.0.0
  * Defeaturing 1.2.1
  * Curves 0.6.21
  * fasteners 0.4.75
  * Render 2023.8.18
```

### Subproject(s) affected?

None

### Anything else?

`Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f80b6533520]
#1  0x7f80b8fe2519 in Gui::DAG::Model::selectionChanged(Gui::SelectionChanges const&) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x509
#2  0x7f80b9252584 in Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x34
#3  /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so(+0x6598fe) [0x7f80b8fd08fe]
#4  0x7f80b92513ef in Gui::SelectionSingleton::slotSelectionChanged(Gui::SelectionChanges const&) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x2cf
#5  /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so(+0x6598fe) [0x7f80b8fd08fe]
#6  0x7f80b925507c in Gui::SelectionSingleton::notify(Gui::SelectionChanges&&) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0xfc
#7  0x7f80b925986f in Gui::SelectionSingleton::setSelection(char const*, std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x83f
#8  0x7f7fda4ca5b7 in FemGui::ViewProviderFemAnalysis::doubleClicked() from /usr/lib/freecad-daily-python3/lib/FemGui.so+0x117
#9  0x7f80b8dbf270 in Gui::Document::slotNewObject(App::DocumentObject const&) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x670
#10  /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so(+0x1a0d9d) [0x7f80b861fd9d]
#11  0x7f80b8612898 in App::Document::addObject(char const*, char const*, bool, char const*, bool) from /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so+0x2c8
#12  0x7f80b868d6d7 in App::DocumentPy::addObject(_object*, _object*) from /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so+0x117
#13  0x7f80b868499c in App::DocumentPy::staticCallback_addObject(_object*, _object*, _object*) from /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so+0x2c
#14  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x128023) [0x7f80b7daf023]
#15  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x8c) [0x7f80b7d66adc]
#16  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9dbc) [0x7f80b7d02a1c]
#17  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7f80b7e493af]
#18  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9d68) [0x7f80b7d029c8]
#19  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7f80b7e493af]
#20  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyEval_EvalCode+0xbe) [0x7f80b7e443de]
#21  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x20c4cd) [0x7f80b7e934cd]
#22  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyRun_StringFlags+0x79) [0x7f80b7e944f9]
#23  0x7f80b8da6e0d in Gui::Application::sDoCommand(_object*, _object*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0xbd
#24  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x128008) [0x7f80b7daf008]
#25  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x8c) [0x7f80b7d66adc]
#26  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9dbc) [0x7f80b7d02a1c]
#27  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7f80b7e493af]
#28  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0xe246c) [0x7f80b7d6946c]
#29  /usr/lib/freecad-daily-python3/lib/libFreeCADBase.so(PP_Run_Method+0xea) [0x7f80b83ef8fa]
#30  0x7f80b83710de in Base::InterpreterSingleton::runMethodVoid(_object*, char const*) from /usr/lib/freecad-daily-python3/lib/libFreeCADBase.so+0x2e
#31  0x7f80b8e3a174 in Gui::PythonCommand::activated(int) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x54
#32  0x7f80b8e314df in Gui::Command::_invoke(int, bool) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x26f
#33  0x7f80b8e31950 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x110
#34  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2f1793) [0x7f80b6c57793]
#35  0x7f80b7708be6 in QAction::triggered(bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x46
#36  0x7f80b770b8fc in QAction::activate(QAction::ActionEvent) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xac
#37  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x26408a) [0x7f80b780708a]
#38  0x7f80b78071e7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xf7
#39  0x7f80b7903d9e in QToolButton::mouseReleaseEvent(QMouseEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xe
#40  0x7f80b77524ee in QWidget::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x20e
#41  0x7f80b770f713 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#42  0x7f80b7717364 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa84
#43  0x7f80b8e02f28 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x98
#44  0x7f80b6c1fe3a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#45  0x7f80b7715e47 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1b7
#46  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c8d40) [0x7f80b776bd40]
#47  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cbfd5) [0x7f80b776efd5]
#48  0x7f80b770f713 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#49  0x7f80b8e02f28 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x98
#50  0x7f80b6c1fe3a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#51  0x7f80b7006307 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x697
#52  0x7f80b6fdba2c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#53  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x73d6e) [0x7f80b0719d6e]
#54  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x26b) [0x7f80b4822d3b]
#55  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xab258) [0x7f80b4878258]
#56  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7f80b48203e3]
#57  0x7f80b6c790b8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x68
#58  0x7f80b6c1e75b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#59  0x7f80b6c26cf4 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#60  0x7f80b8d8deb2 in Gui::Application::runApplication() from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x2822
#61  freecad-daily(+0x6d53) [0x55a574d24d53]
#62  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f80b651ad90]
#63  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f80b651ae40]
#64  freecad-daily(+0x7ea5) [0x55a574d25ea5]`

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarihatari, 2024-01-28:
sorry, tried to format the terminal output better, but didn't succeed..here again:

Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f80b6533520]
#1  0x7f80b8fe2519 in Gui::DAG::Model::selectionChanged(Gui::SelectionChanges const&) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x509
#2  0x7f80b9252584 in Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x34
#3  /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so(+0x6598fe) [0x7f80b8fd08fe]
#4  0x7f80b92513ef in Gui::SelectionSingleton::slotSelectionChanged(Gui::SelectionChanges const&) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x2cf
#5  /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so(+0x6598fe) [0x7f80b8fd08fe]
#6  0x7f80b925507c in Gui::SelectionSingleton::notify(Gui::SelectionChanges&&) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0xfc
#7  0x7f80b925986f in Gui::SelectionSingleton::setSelection(char const*, std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x83f
#8  0x7f7fda4ca5b7 in FemGui::ViewProviderFemAnalysis::doubleClicked() from /usr/lib/freecad-daily-python3/lib/FemGui.so+0x117
#9  0x7f80b8dbf270 in Gui::Document::slotNewObject(App::DocumentObject const&) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x670
#10  /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so(+0x1a0d9d) [0x7f80b861fd9d]
#11  0x7f80b8612898 in App::Document::addObject(char const*, char const*, bool, char const*, bool) from /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so+0x2c8
#12  0x7f80b868d6d7 in App::DocumentPy::addObject(_object*, _object*) from /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so+0x117
#13  0x7f80b868499c in App::DocumentPy::staticCallback_addObject(_object*, _object*, _object*) from /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so+0x2c
#14  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x128023) [0x7f80b7daf023]
#15  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x8c) [0x7f80b7d66adc]
#16  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9dbc) [0x7f80b7d02a1c]
#17  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7f80b7e493af]
#18  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9d68) [0x7f80b7d029c8]
#19  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7f80b7e493af]
#20  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyEval_EvalCode+0xbe) [0x7f80b7e443de]
#21  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x20c4cd) [0x7f80b7e934cd]
#22  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyRun_StringFlags+0x79) [0x7f80b7e944f9]
#23  0x7f80b8da6e0d in Gui::Application::sDoCommand(_object*, _object*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0xbd
#24  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x128008) [0x7f80b7daf008]
#25  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x8c) [0x7f80b7d66adc]
#26  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9dbc) [0x7f80b7d02a1c]
#27  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7f80b7e493af]
#28  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0xe246c) [0x7f80b7d6946c]
#29  /usr/lib/freecad-daily-python3/lib/libFreeCADBase.so(PP_Run_Method+0xea) [0x7f80b83ef8fa]
#30  0x7f80b83710de in Base::InterpreterSingleton::runMethodVoid(_object*, char const*) from /usr/lib/freecad-daily-python3/lib/libFreeCADBase.so+0x2e
#31  0x7f80b8e3a174 in Gui::PythonCommand::activated(int) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x54
#32  0x7f80b8e314df in Gui::Command::_invoke(int, bool) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x26f
#33  0x7f80b8e31950 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x110
#34  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2f1793) [0x7f80b6c57793]
#35  0x7f80b7708be6 in QAction::triggered(bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x46
#36  0x7f80b770b8fc in QAction::activate(QAction::ActionEvent) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xac
#37  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x26408a) [0x7f80b780708a]
#38  0x7f80b78071e7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xf7
#39  0x7f80b7903d9e in QToolButton::mouseReleaseEvent(QMouseEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xe
#40  0x7f80b77524ee in QWidget::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x20e
#41  0x7f80b770f713 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#42  0x7f80b7717364 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa84
#43  0x7f80b8e02f28 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x98
#44  0x7f80b6c1fe3a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#45  0x7f80b7715e47 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1b7
#46  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c8d40) [0x7f80b776bd40]
#47  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cbfd5) [0x7f80b776efd5]
[... cut ...]
#58  0x7f80b6c1e75b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#59  0x7f80b6c26cf4 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#60  0x7f80b8d8deb2 in Gui::Application::runApplication() from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x2822
#61  freecad-daily(+0x6d53) [0x55a574d24d53]
#62  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f80b651ad90]
#63  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f80b651ae40]
#64  freecad-daily(+0x7ea5) [0x55a574d25ea5]

xtemp09, 2024-01-28:
Are the backtraces the same for 0.21._X_? Have you tried _AppImage_? Do other workbenches work?

```cpp
#8 0x7f7fda4ca5b7 in FemGui::ViewProviderFemAnalysis::doubleClicked() from /usr/lib/freecad-daily-python3/lib/FemGui.so+0x117
```

Do you double-click somewhere?

tarihatari, 2024-01-28:
> Have you tried _AppImage_?

Yes. Backtraces for `FreeCAD_weekly-builds-35735-2024-01-26-conda-Linux-x86_64-py310.AppImage` and `FreeCAD_0.21.2-2023-12-17-conda-Linux-x86_64-py310.AppImage` are the same and to me look pretty similar to the one from freecad-daily. (see below)

> Do other workbenches work?

yes (Part, Part Design, Spreadsheet, Sketcher...) always work as expected.

> Do you double-click somewhere?

No. Only one single click on Analysis Container.

> Are the backtraces the same for 0.21._X_? 

`./FreeCAD_0.21.2-2023-12-17-conda-Linux-x86_64-py310.AppImage 
FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.

Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f16b768e520]
#1  0x7f16ba0bf337 in Gui::DAG::Model::selectionChanged(Gui::SelectionChanges const&) from /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libFreeCADGui.so+0x527
#2  0x7f16ba3a15ac in Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&) from /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libFreeCADGui.so+0x2c
#3  /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libFreeCADGui.so(+0x7136ed) [0x7f16ba0936ed]
#4  0x7f16ba3a183f in Gui::SelectionSingleton::slotSelectionChanged(Gui::SelectionChanges const&) from /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libFreeCADGui.so+0x25f
#5  /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libFreeCADGui.so(+0x7136ed) [0x7f16ba0936ed]
#6  0x7f16ba3a25ae in Gui::SelectionSingleton::notify(Gui::SelectionChanges&&) from /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libFreeCADGui.so+0x1be
#7  0x7f16ba3a7362 in Gui::SelectionSingleton::setSelection(char const*, std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&) from /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libFreeCADGui.so+0x5f2
#8  0x7f15d9e8b371 in FemGui::ViewProviderFemAnalysis::doubleClicked() from /tmp/.mount_FreeCAadcozT/usr/lib/FemGui.so+0x121
#9  0x7f16b9e3bade in Gui::Document::slotNewObject(App::DocumentObject const&) from /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libFreeCADGui.so+0x73e
#10  /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libFreeCADApp.so(+0x21449f) [0x7f16b921449f]
#11  0x7f16b922415b in App::Document::addObject(char const*, char const*, bool, char const*, bool) from /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libFreeCADApp.so+0x2bb
#12  0x7f16b92b0ffd in App::DocumentPy::addObject(_object*, _object*) from /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libFreeCADApp.so+0xfd
#13  0x7f16b92b2084 in App::DocumentPy::staticCallback_addObject(_object*, _object*, _object*) from /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libFreeCADApp.so+0x24
#14  /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libpython3.10.so.1.0(+0x148e6d) [0x7f16b972be6d]
#15  /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x27b) [0x7f16b972518b]
#16  /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x5463) [0x7f16b9721013]
#17  /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libpython3.10.so.1.0(_PyFunction_Vectorcall+0x75) [0x7f16b972c305]
#18  /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x4ba8) [0x7f16b9720758]
#19  /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libpython3.10.so.1.0(+0x1df130) [0x7f16b97c2130]
#20  /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libpython3.10.so.1.0(PyEval_EvalCode+0x8c) [0x7f16b97c207c]
#21  /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libpython3.10.so.1.0(+0x20e233) [0x7f16b97f1233]
#22  /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libpython3.10.so.1.0(+0x2095ca) [0x7f16b97ec5ca]
#23  /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libpython3.10.so.1.0(PyRun_StringFlags+0x7d) [0x7f16b97e492d]
#24  0x7f16b9dddf78 in Gui::Application::sDoCommand(_object*, _object*) from /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libFreeCADGui.so+0xb8
#25  /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libpython3.10.so.1.0(+0x148e98) [0x7f16b972be98]
#26  /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x27b) [0x7f16b972518b]
#27  /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x5463) [0x7f16b9721013]
#28  /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libpython3.10.so.1.0(+0x155332) [0x7f16b9738332]
#29  /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libFreeCADBase.so(PP_Run_Method+0xe3) [0x7f16b8f5a5d3]
#30  0x7f16b8eab92b in Base::InterpreterSingleton::runMethodVoid(_object*, char const*) from /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libFreeCADBase.so+0x2b
#31  0x7f16b9ec49b3 in Gui::PythonCommand::activated(int) from /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libFreeCADGui.so+0x53
#32  0x7f16b9ec5108 in Gui::Command::_invoke(int, bool) from /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libFreeCADGui.so+0x2a8
#33  0x7f16b9ec54c4 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libFreeCADGui.so+0x134
#34  /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libQt5Core.so.5(+0x2e9214) [0x7f16b7d7f214]
#35  0x7f16b8869422 in QAction::triggered(bool) from /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libQt5Widgets.so.5+0x32
#36  0x7f16b886bdeb in QAction::activate(QAction::ActionEvent) from /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libQt5Widgets.so.5+0xab
#37  /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libQt5Widgets.so.5(+0x25cb9a) [0x7f16b895bb9a]
[... cut ...]
#59  0x7f16b7d506a2 in QCoreApplication::exec() from /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libQt5Core.so.5+0x82
#60  0x7f16b9ddbddc in Gui::Application::runApplication() from /tmp/.mount_FreeCAadcozT/usr/bin/../lib/libFreeCADGui.so+0x1ecc
#61  /tmp/.mount_FreeCAadcozT/usr/bin/freecad(+0x7620) [0x55622c6a2620]
#62  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f16b7675d90]
#63  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f16b7675e40]
#64  /tmp/.mount_FreeCAadcozT/usr/bin/freecad(+0x7981) [0x55622c6a2981]
`

xtemp09, 2024-01-28:
@tarihatari, what about workbenches utilizing _Python_, like _Arch_, _Path_, _OpenSCAD_?

tarihatari, 2024-01-28:
> @tarihatari, what about workbenches utilizing _Python_, like _Arch_, _Path_, _OpenSCAD_?

no issues...

tarihatari, 2024-01-29:
@xtemp09 I think I found the issue: after compiling FC myself and doing the same steps, I got the following backtrace instead:

`FreeCAD: /home/user/bin/freecad-source/src/Gui/DAGView/DAGModelGraph.cpp:99: const Gui::DAG::GraphLinkRecord& Gui::DAG::findRecord(const string&, const GraphLinkContainer&): Assertion 'it != list.end()' failed.
Aborted (core dumped)`

I actually have the DAGView parameter set to true, because I need to work quite a bit with Part sometimes and when it becomes more complex the DAG view is much more helpful to me than the Tree View. 

Setting DAGView to false "fixes" the symptom temporarily. 

Would be super nice if this could be fixed at some point :)

xtemp09, 2024-01-29:
I've reproduced the crash.

<details><summary>Backtrace</summary>

```cpp

(lldb) bt
* thread #1, name = 'FreeCAD', stop reason = signal SIGABRT
  * frame #0: 0x00007ffff34969fc libc.so.6`__GI___pthread_kill at pthread_kill.c:44:76
    frame #1: 0x00007ffff34969b0 libc.so.6`__GI___pthread_kill [inlined] __pthread_kill_internal(signo=6, threadid=140737259076224) at pthread_kill.c:78:10
    frame #2: 0x00007ffff34969b0 libc.so.6`__GI___pthread_kill(threadid=140737259076224, signo=6) at pthread_kill.c:89:10
    frame #3: 0x00007ffff3442476 libc.so.6`__GI_raise(sig=6) at raise.c:26:13
    frame #4: 0x00007ffff34287f3 libc.so.6`__GI_abort at abort.c:79:7
    frame #5: 0x00007ffff342871b libc.so.6`__assert_fail_base.cold at assert.c:92:3
    frame #6: 0x00007ffff3439e96 libc.so.6`__GI___assert_fail(assertion="it != list.end()", file="/home/xtemp09/Downloads/freecad-source/src/Gui/DAGView/DAGModelGraph.cpp", line=99, function="const Gui::DAG::GraphLinkRecord& Gui::DAG::findRecord(const std::string&, const GraphLinkContainer&)") at assert.c:101:3
    frame #7: 0x00007ffff70fb33e libFreeCADGui.so`Gui::DAG::findRecord(stringIn=<unavailable>, containerIn=0x0000555559330640) at DAGModelGraph.cpp:99:3
    frame #8: 0x00007ffff70ed05f libFreeCADGui.so`operator(__closure=0x00007fffffff9738, in=<unavailable>) at DAGModel.cpp:399:47
    frame #9: 0x00007ffff70ee7ad libFreeCADGui.so`Gui::DAG::Model::selectionChanged(this=0x0000555558c3d8b0, msg=0x0000555555c6a0f0) at DAGModel.cpp:431:36
    frame #10: 0x00007ffff70eaea1 libFreeCADGui.so`Gui::DAG::View::onSelectionChanged(this=0x0000555555c40fb0, msg=0x0000555555c6a0f0) at DAGView.cpp:123:26
    frame #11: 0x00007ffff7383e69 libFreeCADGui.so`Gui::SelectionObserver::_onSelectionChanged(this=<unavailable>, msg=<unavailable>) at Selection.cpp:152:27
    frame #12: 0x00007ffff7392615 libFreeCADGui.so`void std::__invoke_impl<void, void (Gui::SelectionObserver::*&)(Gui::SelectionChanges const&), Gui::SelectionObserver*&, Gui::SelectionChanges const&>((null)=__invoke_memfun_deref @ 0x00007fffffff9a80, __f=<unavailable>, __t=<unavailable>, (null)=<unavailable>) at invoke.h:74:46
    frame #13: 0x00007ffff7392633 libFreeCADGui.so`boost::detail::function::void_function_obj_invoker1<std::_Bind<void (Gui::SelectionObserver::* (Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>, void, Gui::SelectionChanges const&>::invoke(boost::detail::function::function_buffer&, Gui::SelectionChanges const&) [inlined] std::__invoke_result<void (Gui::SelectionObserver::*&)(Gui::SelectionChanges const&), Gui::SelectionObserver*&, Gui::SelectionChanges const&>::type std::__invoke<void (Gui::SelectionObserver::*&)(Gui::SelectionChanges const&), Gui::SelectionObserver*&, Gui::SelectionChanges const&>((null)=0x0000555555c6a0f0, (null)=<unavailable>, __fn=<unavailable>) at invoke.h:96:40
    frame #14: 0x00007ffff739262e libFreeCADGui.so`boost::detail::function::void_function_obj_invoker1<std::_Bind<void (Gui::SelectionObserver::* (Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>, void, Gui::SelectionChanges const&>::invoke(boost::detail::function::function_buffer&, Gui::SelectionChanges const&) [inlined] void std::_Bind<void (Gui::SelectionObserver::* (Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>::__call<void, Gui::SelectionChanges const&, 0ul, 1ul>((null)=<unavailable>, __args=0x00007fffffff9a88, this=<unavailable>) at functional:506:24
    frame #15: 0x00007ffff739262a libFreeCADGui.so`boost::detail::function::void_function_obj_invoker1<std::_Bind<void (Gui::SelectionObserver::* (Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>, void, Gui::SelectionChanges const&>::invoke(boost::detail::function::function_buffer&, Gui::SelectionChanges const&) [inlined] void std::_Bind<void (Gui::SelectionObserver::* (Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>::operator()<Gui::SelectionChanges const&, void>((null)=0x0000555555c6a0f0, this=<unavailable>) at functional:591:32
    frame #16: 0x00007ffff7392626 libFreeCADGui.so`boost::detail::function::void_function_obj_invoker1<std::_Bind<void (Gui::SelectionObserver::* (Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>, void, Gui::SelectionChanges const&>::invoke(function_obj_ptr=<unavailable>, a0=0x0000555555c6a0f0) at function_template.hpp:158:11
    frame #17: 0x00007ffff70e6501 libFreeCADGui.so`boost::function1<void, Gui::SelectionChanges const&>::operator()(this=<unavailable>, a0=<unavailable>) const at function_template.hpp:763:28
    frame #18: 0x00007ffff70e65bf libFreeCADGui.so`boost::signals2::detail::void_type boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type, Gui::SelectionChanges const&>::operator()<boost::shared_ptr<boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int>>, boost::signals2::slot<void (Gui::SelectionChanges const&), boost::function<void (Gui::SelectionChanges const&)>>, boost::signals2::mutex>>>(boost::shared_ptr<boost::signals2::detail::connection_body<std::pair<boost::signals2::detail::slot_meta_group, boost::optional<int>>, boost::signals2::slot<void (Gui::SelectionChanges const&), boost::function<void (Gui::SelectionChanges const&)>>, boost::signals2::mutex>> const&) const [inlined] boost::signals2::detail::void_type boost::signals2::detail::call_with_tuple_args [... cut ...]
    frame #136: 0x00007ffff3429e40 libc.so.6`__libc_start_main_impl(main=(FreeCAD`main at MainGui.cpp:105:1), argc=1, argv=0x00007fffffffddb8, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffdda8) at libc-start.c:392:3
    frame #137: 0x0000555555569c05 FreeCAD`_start + 37
(lldb)

```

</details>

<details><summary>Ubuntu 22.04.3 LTS (KDE/plasma)</summary>

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35733 (Git)
Build type: Debug
Branch: main
Hash: f75db0dbb2f48f3b486afa80afe9d75ab19970c9
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

</details>

commit b8e460f2 (2024-02-12): Fem: Prevent selection of partially attached object - fixes #12163 (#12220)

* Fem: Use of wrong class name in macro
* Fem: Prevent selection of partially attached object - fixes #12163
closed 2024-02-12 by chennes