FreeCAD/FreeCAD #12164: [Problem] Assembly: non-ASCII characters are not always supported

pierreporte, 2024-01-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

First, create a part with PartDesign. Name the Part container with non-ASCII characters like accented letters of Greek letters. Save the file with non-ASCII characters in the name.

Second, create an assembly then use `Assembly_InsertLink` to fetch the part. In the Tasks panel, you’ll notice that the Part container’s name is displayed correctly while the file name in parentheses has two underscores when each non-ASCII characters is supposed to be. After the part is inserted into the assembly, you’ll notice that the Part container’s name has the underscores replacements, while it was displayed correctly in the list.

### Full version info

```shell
0.22 with PR #10764
Tested width Ondsel ES 2024.1
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-02-26:
Somehow createdLink = self.assembly.newObject("App::Link", selectedPart.Label) is failing.
But if you create a Link, then it gets the name correctly. I'll investigate

wwmayer, 2024-02-26:
> Works correctly, but doing this :

The passed string for the internal name is changed to fulfill the rules of a Python identifier and this modified string is also assigned to the Label property.

Anyway, like in RT branch we will make the change to allow any unicode string because since version 3 this is also supported in Python.

PR #12406 (2024-05-06): [Assembly] Add 'Planar' special drag mode for distance joint between 2 planes
- Add a special drag mode for the distance joint when applied between 2 planes.
- Refactor some code so that finding the type of distance joint is externalize.
- Fix shortcut of fixed joint
- Improve tooltip of distance joint.
- Fixes #12452
- Fixes #12457
- Fixes #12544
- Fixes #12550
- Fixes #12551
- Fixes #10750
- Fixes #12164
- Fixes #12449
- Fixes #12168
- Fixes #12165
- Fixes #12419
- Fixes #12445
- Fixes #12440
- Fixes #12889
- Fixes #12887
- Fixes #12890
- Fixes #12884
- Fixes #12752
- Fixes #12656
- Fixes #1 [... cut]
closed 2024-05-06 by yorikvanhavre