FreeCAD/FreeCAD #12165: [Problem] Assembly: the name of `Assembly_InsertLink` is not self-explanatory

pierreporte, 2024-01-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

“Insert link” is what the program does internally to achieve the insertion of objects in the current assembly. The end user wants to insert a part or a sub-assembly. It should be named for what typical users should expect the feature does, not how it does it.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-02-26:
What name do you suggest? "Insert Part" would be kind of misleading imo. Perhaps just "Insert"

pierreporte, 2024-02-26:
The competition uses “insert component”, which suits both parts and sub-assemblies.

PR #12406 (2024-05-06): [Assembly] Add 'Planar' special drag mode for distance joint between 2 planes
- Add a special drag mode for the distance joint when applied between 2 planes.
- Refactor some code so that finding the type of distance joint is externalize.
- Fix shortcut of fixed joint
- Improve tooltip of distance joint.
- Fixes #12452
- Fixes #12457
- Fixes #12544
- Fixes #12550
- Fixes #12551
- Fixes #10750
- Fixes #12164
- Fixes #12449
- Fixes #12168
- Fixes #12165
- Fixes #12419
- Fixes #12445
- Fixes #12440
- Fixes #12889
- Fixes #12887
- Fixes #12890
- Fixes #12884
- Fixes #12752
- Fixes #12656
- Fixes #1 [... cut]
commit f9593d23 (2024-05-06): Assembly: Change 'Insert Link' name to 'Insert Component' Fixes #12165
closed 2024-05-06 by yorikvanhavre