FreeCAD/FreeCAD #12166: [Problem] Assembly: the name of `Assembly_CreateJointFixed` is too long

pierreporte, 2024-01-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The name of `Assembly_CreateJointFixed` is: “If an assembly is active: Create a Fixed Joint. If a part is active: Position sub parts by matching.”

In the tooltip, it needs two lines. A more simple name should be used and the description of what it does should be added to the second part of the tooltip.

It looks to be two features packed into one. It could be split into one that create a joint, working only in assemblies, and a second one to position bodies in parts, working only in a part. The first one would stay in the assembly workbench and the second one would be in part workbenches.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-02-26:
This is fixed in main I think. If not it's fixed in https://github.com/FreeCAD/FreeCAD/pull/12406

pierreporte, 2024-03-08:
@PaddleStroke This is indeed fixed already in the following version:

```
Version: 0.22.0dev.36166 (Git) AppImage
Build type: Release
Branch: main
Hash: a5ea8023a29b6d7bbf1ebd57b5ed31dd27c5947b
```

closed 2024-03-08 by pierreporte