FreeCAD/FreeCAD #12167: [Problem] Assembly: `Std_LinkSelectLinked` and `Std_LinkSelectLinkedFinal` are hard to reach

pierreporte, 2024-01-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the assembly workbench, `Std_LinkSelectLinked` and `Std_LinkSelectLinkedFinal` are used to open a part or sub-assembly in a dedicated tab. This is a common action when working on an assembly. The two commands are located in the Link actions entry of the context menu. Depending on what is right-clicked (tree view entry, a part in the 3D view), it changes position.

The commands should be at the top level of the context menu and in the assembly toolbar.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-02-26:
[image]
Indeed the menu entries are not consistent.

I would suggest moving Std_LinkSelectLinked to the top of the contextual menu when the selection is an external link. I would even say make it the default action of double click (when there is no edit action) rather than having transform as the default.
The contextual menu of tree and objects need some work btw... They are indeed not very consistent of logical.

pierreporte, 2024-02-26:
> I would even say make it the default action of double click

No please. In CAD program, this usually enters the edit mode of the part. I actually asked this for part design but it should be more general. See #11429.

pierreporte, 2024-02-26:
I would add, following the #12165 fix, that the commands should be called “Open component in a new tab” or something similar.

pierreporte, 2024-05-06:
@maxwxyz You may be interested in this since you work on improving context menus.

PR #14032 (2024-05-20): Tree: Add 'Go To Linked Obj' to context menu of Links
Add 'Go To Linked Obj' to Tree context menu of Links.
Fixes #12167

[image]
commit 908bbfeb (2024-05-20): Tree Context menu: add 'go to linked object' action for links. Fixes #12167
closed 2024-05-20 by chennes