FreeCAD/FreeCAD #12169: [Problem] Assembly: the assembly must exist in a file to insert parts in it

pierreporte, 2024-01-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Create an empty file but don’t save it.
2. Create an assembly.
3. Use `Assembly_InsertLink` to insert a part in the assembly.
4. You get the following message: “The current document must be saved before inserting an external part”.

It should be possible so start working without having to save it first.

It is not possible to do quick and temporary assembly tests to evaluate design without having to create a new file first and maybe deleting it after.

Not a big deal but a little annoying.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-01-30:
I think this is a limitation of App::Link. You can't have external links without saving the file first if I'm not mistaken

pierreporte, 2024-03-10:
@realthunder Do you confirm that the issue is caused by the implementation of links?

PaddleStroke, 2024-03-11:
@pierreporte you can just try to do the same with a document without an assembly. Creating a link to an external object in an unsaved file gives a 'owner not saved' error

pierreporte, 2024-03-11:
@maxwxyz can you change the label of the issue?

realthunder, 2024-03-31:
Please reconsider #11037 which not only fixes the issue here, but also has much more potential usage in the context of assembly management. For example, allow user to build parts library that can be shared among unrelated assemblies.