FreeCAD/FreeCAD #12170: [Problem] Double click opening a FreeCAD file in MacOS does not result in the file being the first to display

bdlabitt, 2024-01-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With FreeCAD unopened, a double click of a FreeCAD file results in the Startup page being in the foreground, rather than the file that was opened.  This is not expected behavior for an application, be it Windows, Linux or MacOS.  The double clicked file has been opened but is in the background.

### Full version info

```shell
[code]
OS: macOS 14.2.1
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * A2plus 0.4.62
  * fasteners 0.5.0
  * Help 1.0.3
[/code]
Also confirmed by @crisb with 0.22dev in a reply on the FreeCAD users forum.  Chrisb recommended I file this issue.
```

### Subproject(s) affected?

None

### Anything else?

[image]
I use 0.21.1 as my default, but I can and have built 0.22dev.  I have not installed 0.22dev to the system, but can run it via the command line.  Chrisb reports similar behavior for 0.22dev.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bdlabitt, 2024-01-28:
Where I wrote crisb, I meant chrisb.  Sorry about that typo.

JohnOCFII, 2024-01-28:
I can confirm this behavior still exists as of the 35720 build.  For the screen shot below, I double-clicked on the param_rect.FCstd file in the macOS Finder and while the file is properly opened, the Start Page is shown as the top-most window.  This behavior seems to happen regardless of the `Switch workbench after loading` setting.

[image]

```
OS: macOS 14.3
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35720 (Git)
Build type: Release
Branch: main
Hash: 00cd50bbf49c810555899efe112ec7595c76cd3a
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * OpenLight 2023.12.17
  * FeedsAndSpeeds 0.5.0
  * OpenDark 2023.12.17
```
[image]

luzpaz, 2024-08-24:
@JohnOCFII still reproducible in latest HEAD ?

JohnOCFII, 2024-08-24:
> @JohnOCFII still reproducible in latest HEAD ?

Nope.  Problem seems resolved with recent builds of 0.22. (Tested with 38459, as that is the last weekly app bundle I have due to the Mac packaging issues for Arm Macs.)

closed 2024-08-24 by luzpaz