FreeCAD/FreeCAD #12173: [Problem] Copy-pasting an exponentiated value with a unit in spreadsheet changes the value.

markasoftware, 2024-01-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

To reproduce:

1. Create new spreadsheet.
2. Create cell with value `=(5mm)^2`. The value will appear as `25.00 mm^2` when rendered.
3. Copy-paste the cell to another cell.
4. Observe that the formula for the new cell is `=5(mm ^ 2)`, and is rendered as `5.00 mm^2`

### Full version info

```shell
[code]
OS: Debian GNU/Linux trixie/sid (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.20.2.
Build type: Release
Python 3.11.5, Qt 5.15.10, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Spreadsheet

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-24:
@markasoftware can you test with dev build ?

MiqCG, 2024-08-26:
Hi,

Still present in revision 38459.
Looks like the brackets get moved from `=5 mm ^ 2` to `=5 (mm ^ 2)` on copy-paste. The problem appears if you copy-paste the cell or the "Content" at the menu bar.

[image]

OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git) AppImage
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * Movie 2024.2.15
  * sheetmetal 0.4.24
  * FreeCAD-motion-workbench-master
  * dodo 1.0.1
  * PieMenu 1.2.7
  * fasteners 0.5.27

MiqCG, 2025-02-12:
There are still problems with this, even worse. Now I have lost the possibility to indicate the exponent "²" using the "^" keystroke. It's weird because I can not exponentiate any number in any version of FreeCAD but the keystroke works and creates an exponent in other programs (for example LibreOffice). 
Entering the sequence "=5^2" results in "52". Entering "=(5mm)^2" results in "'=(5mm)2"

In revisions 40122 and 1.0.0 and 37543 (pre-1.0), the report view says:
 <Spreadsheet> Cell.cpp(676): Unnamed#Spreadsheet.A2: Failed to parse expression.

`
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.40122 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 103/100.707
`

xtemp09, 2025-02-13:
I cannot reproduce. Though, when pressing "2", the cell editing finishes, but it is a known issue.

@MiqCG, are you experiencing problem when entering "^", as reported in #17738? Are you using QWERTZ keyboard?

```ruby
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40176 (Git) Conda
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/102.706
```

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40233 (Git)
Build type: Debug
Branch: master
Hash: 8899e3476e770f3d1da5099683a4a167b21fd0b1
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96.0584/96.0584
```

MiqCG, 2025-02-13:
Didn't know about this issue, @xtemp09 . Thanks. 
I am not with a QWERTZ keyboard but a QWERTY one. I have reported my situation in there:
https://github.com/FreeCAD/FreeCAD/issues/17738#issuecomment-2656655120

So, in relation to this issue, my situation doesn't count.

luzpaz, 2025-06-13:
@MiqCG JFYI markdown codeblocks are surrounded by 3 backticks

```
This is a successful codeblock
```

`
This isn't a codeblock
`

xtemp09, 2025-06-27:
1. Create a cube, enter an expression into, for example, _Length_: `(5 mm) ^ 2 / 1 mm`.
2. Observe `5 (mm ^ 2) / 1 mm`
3. Copy and paste it. See the same `5 (mm ^ 2) / 1 mm`

The issue is not about _Spreadsheet_, but about the expression mechanism. Please, remove the tag _Mod: Spreadsheet_.

I think that this issue is a **blocker**. Is my understanding correct that it all began with PRs "_Units cleanup_", "_Units rework II_"? 

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.42340 (Git)
Build date: 2025/06/27 09:25:36
Build type: Unknown
Branch: master
Hash: 4f167b828b68b1544732881f0b2393ea873cffd1
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.13
shiboken 5.15.13, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.6.3
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/unset/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96.2463/96.2463/1
```

JohnOCFII, 2025-12-11:
> * Create a cube, enter an expression into, for example, _Length_: `(5 mm) ^ 2 / 1 mm`.
> * Observe `5 (mm ^ 2) / 1 mm`
> * Copy and paste it. See the same `5 (mm ^ 2) / 1 mm`

Interesting.  When I make expressions, I've never used measurement units.  So, `5 ^ 2/1` works just fine with cut and paste.

I did just try this with the expression: `(7 mm) ^ 2 / 1 mm` and I was able to confirm the issue, as the result (after pasting, closing, then re-opening the pasted expression) was: `7 (mm ^ 2) / 1 mm`

```
OS: macOS 26.1
Architecture: arm64
Version: 1.2.0dev.44597 (Git)
Build date: 2025/12/10 11:42:13
Build type: Unknown
Branch: main
Hash: 67948d60a26c790dc8f233489134ab588dea6b2f
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2025.5.20
  * Render 2024.12.15
  * Telemetry 1.0.4
  * Curves 0.6.70
```

xtemp09, 2025-12-14:
Ok. The issue is present for years. Here is the list of the versions I tried.

<details>

```ruby
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +99 (Git)
Build type: Release
Branch: Branch_0.19.3
Hash: 6530e364184ce05ccff39501e175cf2237e6ee4b
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: Russian/Russia (ru_RU)
```

```ruby
OS: Ubuntu 24.04.3 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

```ruby
OS: Ubuntu 24.04.3 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0rc1.20251125 (Git shallow) AppImage
Build date: 2025/11/25 17:53:08
Build type: Release
Branch: (HEAD detached at 5376760)
Hash: 53767601ea65de7cd9f6590e7d5856afc0841635
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/96.0584/1
```

</details>

I still think it must be solved, since people will laugh when they find out that _FreeCAD_ transforms `(7 mm) ^ 2 / 1 mm` into `7 mm`.

kadet1090, 2025-12-14:
> I still think it must be solved, since people will laugh when they find out that _FreeCAD_ transforms `(7 mm) ^ 2 / 1 mm` into `7 mm`.

Solved sure - I'll leave it with priority high but not regression (since it is not one) nor blocker.

maxwxyz, 2026-11-07:
The same fix as https://github.com/FreeCAD/FreeCAD/pull/30256 is also available in AstoCAD/FreeCAD: https://github.com/AstoCAD/FreeCAD/commit/f5ad02801cec726e0de5684749bb5a79268a623b.