FreeCAD/FreeCAD #12175: [Problem] Base angle doesn't work when using input field in Draft Rotate

FEA-eng, 2024-01-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Draft Rotate has the Base angle setting. However, it works only when the tool is used by selecting the points. When the base angle is specified in the input field, it has no effect on rotation. This makes the base angle setting even more confusing.

Forum discussion: https://forum.freecad.org/viewtopic.php?t=69525

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35691 (Git)
Build type: Release
Branch: main
Hash: cc45b58dadae53d7a99be1ed7db25f6c93bb13d5
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-02-27:
The base angle indeed only makes sense if you are visually defining an angle. Basically it allows you to do "I want to rotate from here to there" which is very useful when you want to align slightly unaligned things. This is useful because it allows you to perform a rotation without knowing the value of the rotation.

When working with the keyboard only, the base angle indeed has no meaning, because you already know of how much you want to rotate, and you of course input the correct value.

I am not convinced by @Roy-043 's PR because it basically removes the scenario I described above. In that case, there would be no point anymore in having the base angle.

So I'm not sure what's the best way to solve this... Maybe rename "Base angle" to something like "Reference angle"? Or to "Base angle (optional)"?

FEA-eng, 2024-02-27:
> The base angle indeed only makes sense if you are visually defining an angle. Basically it allows you to do "I want to rotate from here to there" which is very useful when you want to align slightly [...]

Ok, this definitely makes sense and can indeed be useful.

> So I'm not sure what's the best way to solve this... Maybe rename "Base angle" to something like "Reference angle"? Or to "Base angle (optional)"?

Renaming to "Reference angle" sounds like a good idea but there will still be an issue that the input field for this angle ignores keyboard inputs. Maybe the best way to avoid the confusion would be to gray out this input field so that it only displays the current angle when moving the cursor but doesn’t allow any keyboard inputs since they don’t make sense anyway.

Roy-043, 2024-02-28:
> I am not convinced by @Roy-043 's PR because it basically turns the scenario I described above impossible. And in that case, there would be no point anymore in having the base angle.

@yorikvanhavre That is not correct. As mentioned in the PR:
> The behavior of the command when clicking points in the 3D view remains the same.

Can you do some more testing please?

yorikvanhavre, 2024-02-28:
> @yorikvanhavre That is not correct.
Yes, sorry I wrote wrongly. You can still do that, but the reading you get in the UI is counter-intuitive (at least for me, this is how it works in autocad, basically).

Roy-043, 2024-07-25:
Let's close this.

closed 2024-07-25 by Roy-043