FreeCAD/FreeCAD #12176: [Problem] Arc shape in linear pattern corrupted / twisted

pmainvil, 2024-01-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

See attached image. Discovered the problem when exporting to .STEP and .STL format for 3D print.  Upon closer inspection is visible on the display within FreeCAD 0.21.2

See ZIP File for FreeCAD, STEP and STL files.

Seems to appear if number of slots = 4, but not for other numbers (in Spreadsheet). 

[image]

[image]

[PorteJetonV4.zip](https://github.com/FreeCAD/FreeCAD/files/14077421/PorteJetonV4.zip)

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: French/Canada (fr_CA)
Installed mods: 
  * Assembly4 0.50.6
  * Curves 0.6.22
  * Help 1.0.3
  * Manipulator 1.5.7
  * Pyramids-and-Polyhedrons
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2025-02-10:
@pmainvil would you mind trying to reproduce on v1.0 ?

pmainvil, 2025-02-10:
I have reloaded the file into V1.1.0 dev.  It proceeded with a conversion and saved the file.  The artefact is still present in the FreeCAD render, but the .STEP file exported does not show the problem when Sliced in OrcaSlicer 2.2.0

When I reimported the .STEP file into FreeCAD 1.1.0dev the result render was OK.

When closing the file and reloading, the visual artefact is still present though.

[image]

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40122 (Git) Conda
Build type: Release
Branch: main
Hash: c1dddaeadc68cc9f05f1312dc1c6789820cfdd5d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell /.0.0, OCC 7.8.1
Locale: French/Canada (fr_CA)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/81.5973
Installed mods: 
  * Curves 0.6.53
  * Defeaturing 1.2.3
  * fasteners 0.5.34
  * FreeCAD-themes 2024.7.24
  * Gridfinity 0.6.0
  * OpenTheme 2024.9.1
  * sheetmetal 0.6.13

pmainvil, 2025-02-10:
Another view exposing the warping better.

[image]

pmainvil, 2025-06-11:
Same results as before.  

[image]

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42133 (Git)
Build date: 2025/06/09 17:51:59
Build type: Release
Branch: main
Hash: 12b4610b499a9756f19bbb85f59fe16e5fa135db
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: French/Canada (fr_CA)
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * Curves 0.6.62
  * Defeaturing 1.2.6
  * fasteners 0.5.38
  * fasteners.backup1748094423.5998592 0.5.38 (Disabled)
  * FreeCAD-themes 2025.1.7
  * FusedFilamentDesign 0.25.200
  * Gridfinity 0.11.9
  * OpenTheme 2025.5.20

FlachyJoe, 2025-06-11:
It's a tessellation problem and it only occurs with some values of Angular deflection and Deviation

[image]

FEA-eng, 2025-09-20:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FEA-eng, 2025-11-21:
> this requires further investigation to exonerate FreeCAD code

Then I guess it should stay open (despite the workaround).

ElementW, 2026-09-15:
> luzpaz OCC tessellation is post-processed for Coin, so this requires further investigation to exonerate FreeCAD code.

Opening the STEP or a BREP export in [Mayo](https://github.com/fougue/mayo), a CAD viewer based on OCC, shows the same problem, so FreeCAD is not to blame here.

[image]

<br>Of a few online viewers, [3dviewer.net](https://3dviewer.net) and [stepviewer.app](https://stepviewer.app) display the same kind of problem (because they're based on a JS transpile of OCC), while [partwork.ai/step-viewer](https://www.partwork.ai/step-viewer/), [imagetostl.com](https://imagetostl.com/view-stp-online) or [ShareCad](https://sharecad.com/step-viewer-online) don't, though none of the latter have any transparency on what they're based on.

Notably though [Autodesk's viewer (link to specific part)](https://viewer.autodesk.com/id/dXJuOmFkc2sub2JqZWN0czpvcy5vYmplY3Q6YTM2MHZpZXdlci1wcm90ZWN0ZWQvdDE3ODk0MzM4MTdfZTg5Mzg5ODAtMzliNi00YzgwLTg2MWMtZmMyZGNkNTZjYjM2LnN0ZXA?sheetId=NzgxYjUyM2EtODMyZS00YjIwLTk1ZDktZTRhMWMwNGI3YzI0) does fine so presumably the STEP itself is okay.

[image]

luzpaz, 2026-09-15:
Using Mayo is a great way to double check some of these OCC bugs in that case. Thanks for the tip!