FreeCAD/FreeCAD #12178: [Problem] Dimensions, radiuses and other things are displayed as cut in half

skrz1, 2024-01-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the techdraw, all of the informations of the dimensios are cut in half. The problem is even worse after exporting to PDF.
[image]

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
[/code]
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-02-18:
Need a font and a FCStd file before I can tell you much.

luzpaz, 2024-03-13:
@skrz1 :point_up_2:

maxwxyz, 2024-05-25:
I'll close this, no info for over two months.
@skrz1 you can reopen if this issue is still relevant you provide additional info.

closed 2024-05-25 by maxwxyz