FreeCAD/FreeCAD #12191: [Feature Request] Add constrain editing to Dimension tool

pike2k, 2024-01-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using the Dimension tool, currently you must exit tool in order to edit constraints. It would improve workflow if it would be possible to also edit existing constraints in sketch by clicking on them with Dimension tool active.

### Full version info

```shell
0.22.0Dev
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-08-30:
@maxwxyz This is UI/UX, not CAD feature

maxwxyz, 2024-08-30:
@pierreporte agreed, you've added this :)