FreeCAD/FreeCAD #12193: [Problem] When inputing an expression in attachment offset angle, if it's result is > 180° or < -180° it gives an error

Anthony-Gaudino, 2024-01-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the Attachment panel, in the Attachment Offset section, if I enter an expression which results in a value that is > 180° or < -180° FreeCAD shows this error: `<Exception> Roll angle is out of range [-180, +180]` but it does rotate to the desired location.

For example, if I input `45 ° + 180 °` it rotates to `-135°`, but gives the error.

I expect FreeCAD to accept values higher than 180° or less than -180° and calculate the correct rotation and don't display the error.

### Full version info

```shell
OS: macOS 14.3
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.9.22
  * ThreadProfile 1.89.0
  * Launcher
  * InventorLoader 1.4.0
  * woodworking 0.21.1.33694
  * Render 2024.1.23
  * Silk 0.1.5
  * Defeaturing 1.2.2
  * A2plus 0.4.62
  * Manipulator 1.5.7
  * fasteners 0.5.0
  * lattice2 1.0.0
  * CurvedShapes 1.0.5
  * sheetmetal 0.3.16
  * toSketch 1.0.1
  * Curves 0.6.22
  * Assembly4 0.50.6
  * Assembly3 0.12.2
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pitch-circle, 2024-02-17:
May I ask why this issue has an Expressions label?

In a way, Expressions are the solution/workaround to this issue not the problem. With the remainder and mod() operators, you can ensure that your input figure falls within the currently-allowable range. 

Hence, it would be possible to argue that this is not a bug but a documentation problem.  There are other places in the documentation that specify the acceptable range for a particular angular value as between -180 and +180 (I think the documentation says strictly greater than -180 and strictly less than +180). Add the same words to the Wiki page for this item and the issue is resolved without any code changes.

luzpaz, 2024-03-09:
Thanks for the feedback. 
To answer your question, perhaps it may have an expression element to it? A label doesn't only connote the cause.

closed 2025-06-25 by github-actions[bot]