FreeCAD/FreeCAD #12198: [Problem] TechDraw Tool Baloon

sebaras, 2024-01-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

To define a surface, Baloon A should look like Baloon B (line in center from Baloon)
The position of the line, should be by the red line

### Full version info

```shell
[code]
OS: Windows 10 build 19042
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Switzerland (de_CH)
Installed mods: 
  * A2plus 0.4.60n
  * Assembly4 0.50.2
  * CADExchanger
  * ExplodedAssembly
  * fasteners 0.4.60
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

Techdraw

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

benj5378, 2024-08-30:
How should this be fixed? Automatically snapping to whatever side isn't preferred either.

Perhaps 3 radio buttons: Anchor, horizontal, vertical horizontal/vertical? Any inputs? @WandererFan

luzpaz, 2024-09-12:
@WandererFan :point_up_2:

luzpaz, 2024-09-12:
Remotely related: #16443

FEA-eng, 2026-08-15:
[image]

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 26.3.0dev.20260812 (Git shallow)
Build date: 2026/08/12 05:41:29
Build type: Release
Branch: (HEAD detached at dc1ef6e88)
Hash: dc1ef6e880d6f5cf5d5a6eb3749373141eb0deb8
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 610.88/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```