FreeCAD/FreeCAD #12199: [Problem] (qt.qpa.fonts) Populating font family aliases took 142 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost.

xgpt, 2024-01-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When opening FreeCAD (and also Ondsel), error is thrown when opening on macOS Monterey Arm64

Not sure what font family issue is taking place, but this is an error each and every time I open FreeCAD on Mac

### Full version info

```shell
OS: macOS 12.7.2
Word size of Ondsel: 64-bit
Version: 2024.1.0.35694 (Git)
Build type: Release
Branch: flavor
Hash: b070090c6873b3fcf4be9a459219d22edfc4e7cf
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Ondsel-Lens 2024.1.22.01
  * OpenDark 2023.12.17
```

Also affected in latest FreeCAD build via Brew. I just happen to have the Ondsel build up and running right now.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

JohnOCFII, 2024-01-31:
Are you referring to this message:

`21:01:44  (qt.qpa.fonts) Populating font family aliases took 51 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost. `

If so -- from what I understand, that error is coming from the underlying QT libraries.  I'm not sure if there is anything FreeCAD can do (other than perhaps note it somewhere).

```
OS: macOS 14.3
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35759 (Git)
Build type: Release
Branch: main
Hash: 90d36f60c30cb6549a7987fcd9d4769eab1c160c
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * OpenLight_Beta 2023.12.17
  * FeedsAndSpeeds 0.5.0
  * OpenDark 2023.12.17
```

xgpt, 2024-02-01:
I believe the reason the QT library is calling that error is because FreeCAD is calling for a specific non-existent font instead of calling for a font-family that is guaranteed to be on the platform. If that's so, that *is* something that I think can be fixed in FreeCAD. I'm not sure who to speak to about this who knows the codebase better.

JohnOCFII, 2024-02-01:
@obelisk79 -- Any thoughts on this?

obelisk79, 2024-02-01:
I'm not entirely sure what the cause is, or best solution should be. Instinctively I imagine there should be an easy way to explicitly pull and use the system font on whatever platform FreeCAD is being run on. Otherwise the next best solution would be to package and use an opensource font file and avoid some of these cross platform issues. As far as the message itself goes, it can be safely ignored entirely. I have the report view completely disabled, to me it seems to serve better as a debug console than anything useful to the average user.

Syres916, 2024-02-03:
@JohnOCFII Courier is specified in at least three places in FreeCAD's code for both _Report View_ and _Editor_ as it appears to be the fallback font if a _monospace_ one cannot be found at the first attempt. Therefore could you please volunteer a default monospace fallback font for Mac's (it has to appear in Preferences>Python>Editor>Family combobox) and I can test some code changes as I'm already looking at eliminating Bar Code fonts from the lookup as they are generating issues mainly with Windows users.

JohnOCFII, 2024-02-03:
> Therefore could you please volunteer a default monospace fallback font for Mac's (it has to appear in Preferences>Python>Editor>Family combobox)

May I suggest `Courier New`?  It shows up as `Courier New` in the Preferences for Python.

It is called `Courier New Version 5.00.2x` in the Font List from Apple.  It is listed as OpenType/TrueType format.

The actual file is: /System/Library/Fonts/Supplemental/Courier New.ttf (note the space).
[image]

xgpt, 2024-02-06:
I tried swapping Courier with both Courier New and also a programming font that I use on my system, I restarted each time, this did not solve the problem.

Syres916, 2024-02-06:
@xgpt please clarify where you changed Courier to Courier New, just in the Preferences or did you actually change the code and recompile, there is a big difference?

luzpaz, 2024-03-06:
@xgpt ping

xgpt, 2024-03-06:
Just in the preferences sorry @luzpaz @Syres916

spkane, 2024-06-09:
I have the exact same issue. It does not seem possible to fix it via the preferences that everyone keeps mentioning. I'd be happy to try and do a little debugging for a FreeCAD dev, if someone wants to tell me what to test and what logs to post.

sslupsky, 2024-09-16:
Problem still exists in RC1.

xgpt, 2024-09-30:
My mac totally fried itself to dead, I purchase a [frame.work](https://frame.work/) laptop and I'm loving it. As I don't think I'll be getting a macos device again for a long time if ever again...I'm going to unsubscribe from this issue. If an admin wants to close it that's fine with me, but I can't contribute any more bug reports personally.

luzpaz, 2024-10-16:
Related:  
* #17275

JohnOCFII, 2024-10-16:
> Related:
> 
> * [Use Qt built-in to determine the system's preferred fixed-width font #17275](https://github.com/FreeCAD/FreeCAD/pull/17275)

Yes -- noticed that.  I hope to test that PR on a self-built FreeCAD today.  Would be great if it also solved this little qt.qpa.fonts message.

brianredbeard, 2024-12-10:
Still occurring in `1.1.0dev.39560`:

```
09:56:20  (qt.qpa.fonts) Populating font family aliases took 125 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost.
```

Also, I can confirm that all of the fonts in the various preferences are set to "Menlo", as can be seen in the second screen shot.

```
OS: macOS 15.1.1
Architecture: arm64
Version: 1.1.0dev.39560 (Git) Conda
Build type: Release
Branch: main
Hash: 1c35d08d87bb6dec7cb5adf51b2ed460ce43096d
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Dracula 0.0.5
  * FreeGrid 2.0.1
  * boltsfc 2022.11.5
  * sheetmetal 0.5.3
```

[image]

[image]

risacher, 2025-01-10:
FWIW, using FreeCAD 1.1.0dev-39786, I no longer get this error for "Courier" but instead I get it for "Sans-serif".

schuessm, 2025-01-28:
I searched the Internet for [Courier.TTF](https://fontsfree.net/courier-font-download.html) and install it. Then I switched in the Python Setting in FreeCAD preference to "Courier" an start it new, now it works. Perhaps it helps you, who have problems on MAC. Oh and before I [deleted](https://wiki.freecad.org/Installing_on_Mac#Uninstallation) all FreeCAD folders on the Mac.

spkane, 2025-04-19:
Although annoying, I finally removed these font errors from my copies of both **FreeCAD** and **AstoCAD**.

| I am pretty confident that this can be fixed in the FreeCAD codebase so that the list of default fonts to load includes ones that are standard on each major OS.

- In **FreeCAD**, I installed [Courier.TTF](https://fontsfree.net/courier-font-download.html), which fixed the first "error." 
- Then I also started getting `(qt.qpa.fonts) Populating font family aliases took 163 ms. Replace uses of missing font family "Sans" with one that exists to avoid this cost.` Although there are many *Sans fonts in the world, I can't find one called Sans that will make this error disappear, so I decided to try and rename one of the many available Open Sans fonts.
  - I downloaded and extracted  [Google OpenSans](https://fonts.google.com/specimen/Open+Sans).
  - I then opened a terminal and changed it to the directory that contained `OpenSans-Regular.ttf`.
  - On macOS, I installed [fontools via HomeBrew](https://formulae.brew.sh/formula/fonttools).
  - Then I created a script named `rename_font.py` based on [an answer from Stack Exchange](https://superuser.com/a/694452/104494).
  ```py
  #!/usr/bin/env python

  from fontTools.ttLib import TTFont
  import sys
  
  def rename_font(input_path, output_path, new_name):
      font = TTFont(input_path)
      for record in font['name'].names:
          if record.nameID in [1, 4, 6]:  # Family name, Full name, PostScript name
              record.string = new_name.encode(record.getEncoding())
      font.save(output_path)

  if __name__ == "__main__":
      if len(sys.argv) != 4:
          print("Usage: rename_font.py <input_path> <output_path> <new_name>")
          sys.exit(1)
      rename_font(sys.argv[1], sys.argv[2], sys.argv[3])
  ```
  - I ran `chmod u+rwx ./rename_font.py` and the ran `./rename_font.py ./OpenSans-Regular.ttf ./Sans-Regular.ttf Sans`
  - After this, I installed `Sans-Regular.ttf` on my system via FontBook.
- In **AstoCAD**, I also got an "error" about **Latin Modern Roman**, so I downloaded and installed the [Latin Modern Roman OTF files from here](https://www.fontget.com/font/latin-modern-roman-family/).

Once I had done all of this and restarted the applications, there were no more font warnings/errors on startup!

luzpaz, 2025-06-02:
See #21710 (although the solution is to blacklist the font) and @spkane found that simply installing the missing  font or renaming the font.

JohnOCFII, 2025-06-02:
> See [#21710](https://github.com/FreeCAD/FreeCAD/pull/21710) (although the solution is to blacklist the font) and [@spkane](https://github.com/spkane) found that simply installing the missing font or renaming the font.

On macOS, the font isn't missing -- I think it is that macOS changed to TTC files a while back, and FreeCAD is still only looking for TTF files.  https://docs.fileformat.com/font/ttc/ At some point (maybe when laid off or retired) I'd like to test this.  https://doc.qt.io/qt-6/qfont.html

leematos, 2025-07-05:
Ok, whew, got a build environment in a VM. Pixi _is_ the easiest way to get that going.

In my playing around, I've gotten the error to change to `Replace uses of missing font family "Latin Modern Roman" with one that exists to avoid this cost` so that at least feels like progress?

Looking at the fonts folder on Mac I attempted both Menlo and Monaco and both seem to end up throwing that error. I need to learn a little bit more about QTs font class to see if I'm accidentally hitting a fallback font.

Interestingly, Menlo is now `Menlow.ttc` but Monaco is still `Monaco.ttf`.

I also quickly tried Courier New, but I didn't change it _everywhere_. Setting up a more robust env now so I can do some better debugging.

JohnOCFII, 2025-07-06:
@leematos From the little bit of testing I've done, it seems like FreeCAD is using some older QT font calls that no longer work to find fonts on modern macOS.  I'd guess the QT font calls need to change -- but not sure **what** exactly needs to change.

leematos, 2025-07-06:
If there are any C++ debug wizards following I could use some help.

I'm trying to make sure QT is in debug https://doc.qt.io/qt-6/debug.html#debugging-in-macos-and-xcode and the layers of build systems are causing me some confusion. No errors yet, as I'm not even sure where to start.

My goal was to step through lldb breakpoints on QFont calls and see what fonts are actually being loaded where, but I'm not getting line by line code output which leads me to believe I don't have a debug build. (lldb newbie too!)

leematos, 2025-07-06:
I mean... I did not want to believe this to be true, but on my system right now, the "Latin Modern Roman" type error is coming from this _stylesheet in this SVG icon_ 🤯:

https://github.com/FreeCAD/FreeCAD/blob/db11f481e9ac74c8f39c513a5cc233058af20fb1/src/Gui/Icons/VarSet.svg#L228

I've removed that line and the Latin Modern Roman error is gone. 

My build is wonky so I need to clean that up -- but possibly solved?! 💪

leematos, 2025-07-06:
Cleaned up my build and No error! 🎉 I'll get something pushed up later.