FreeCAD/FreeCAD #12208: [Problem] Axis colors are not unified and can't be configured

qewer33, 2024-01-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Axis colors (X: Red, Y: Green, Z: Blue) are used it a lot of different places (Sketcher, navgiation cube, axis cross, transform dragger etc.) in FreeCAD but they're usually hardcoded meaning they aren't unified and can't be configured by the user or preference packs/themes.

### Full version info

```shell
OS: Fedora Linux 38 (KDE Plasma) (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35726 (Git)
Build type: Release
Branch: master
Hash: 9f84fbe9bbd476209ad93eedea76d49aa8ce75c7
Python 3.11.7, Qt 5.15.11, Coin 4.0.1, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4.backup1691951820.225657 0.50.2
  * Assembly4
  * Behave-Dark-Colors 0.1.1
  * Curves 0.6.21
  * DynamicData 2.60.0
  * fasteners 0.4.75
  * Freecad-Built-in-themes-beta 1.2.2
  * freecad.gears.backup1692223246.5590403 1.0.0
  * freecad.gears
  * ExtremeProLight 1.6.0
  * A2plus 0.4.61
  * ExplodedAssembly
  * Glass (Disabled)
  * kicadStepUpMod 10.22.8
  * Silk 0.1.5
  * lattice2 1.0.0
  * Manipulator 1.5.6
  * Leaf Dark Colors 0.1.1
  * FreeCAD_DarkPreferencePack 0.0.1
  * OpenDark 2023.12.17
```
```

### Subproject(s) affected?

Core

### Anything else?

In my transform dragger PR (https://github.com/FreeCAD/FreeCAD/pull/10706) I've added axis colors as parameters to the `ViewParams` class
https://github.com/FreeCAD/FreeCAD/blob/206711b8fe8fdbb66e49b20a96f274abf81fb8dd/src/Gui/ViewParams.h#L66-L68
these are currently only used by the transform dragger and can only be changed via the parameter editor. The "groundwork" is already here but what needs to be done is:
- Determine and create a list of the places where axis colors are used across FreeCAD
- Open PRs to make those places use these parameters one by one
- Expose these parameters to the user via `Preferences > Color`. Or they could stay as parameters so only pref packs/themes can change them but not the user. This can be discussed.

Would appreciate help from anyone interested in this.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

obelisk79, 2024-01-31:
This would be of benefit to theme developers

maxwxyz, 2024-04-13:
related: 
- https://github.com/FreeCAD/FreeCAD/issues/12584

luzpaz, 2024-04-13:
Related to https://github.com/FreeCAD/FreeCAD/issues/6805

yorikvanhavre, 2024-04-16:
The Draft grid uses axes colors too. We could also make it use the pref values above... Doing a PR for that

luzpaz, 2024-07-26:
Is this ticket still valid ?

maxwxyz, 2024-07-26:
I think this is only solved for Draft.

luzpaz, 2024-07-27:
@maxwxyz should we address this pre- or post-1.0 ?

maxwxyz, 2024-07-27:
Seems to be no small PR and a new feature, so post 1.0 IMO.

PR #16995 (2025-02-03): [Gui] Axis colors changeable
Partiarlly fixes #12208

Make axis colors user definable
closed 2025-02-03 by chennes