FreeCAD/FreeCAD #12211: [Problem] When placing a Door, a Window get placed instead

nomennescio, 2024-01-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When trying to place a Door (BIM_Door), a Window get placed instead.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * BIM 2021.12.0
  * Help 1.0.3
  * parts_library
  * Reporting
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-01-31:
The Door tool just uses and stores a separate preset compared to the Windows tool, but that preset can be anything. The initial preset was confusing (not a door) but that issue has already been fixed (#11940). Closing this as completed.

closed 2024-01-31 by Roy-043

nomennescio, 2024-01-31:
> The Door tool just uses and stores a separate preset compared to the Windows tool, but that preset can be anything. The initial preset was confusing (not a door) but that issue has already been fixed (#11940). Closing this as completed.

Other than getting a development release, what is the workaround for this? I cannot place doors now.

Roy-043, 2024-01-31:
Select a different option in the task panel, "Simple door" for example.

nomennescio, 2024-01-31:
> Select a different option in the task panel, "Simple door" for example.

Right. Is it possible to change this default somewhere as a preference?

Roy-043, 2024-01-31:
You need to get better acquainted with the tool first. Please use the [forum](https://forum.freecad.org) if you have any remaining questions.