FreeCAD/FreeCAD #12212: [Problem] When Undo-ing a place Door (BIM_Door), hole in wall persists

nomennescio, 2024-01-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I try to place a door (BIM_Door) in a Wall, but am not satisfied, and do an Undo, the door is removed, but the hole in the wall persists. If I save the file, and reopen, the hole is still there, meaning it is in the saved file.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * BIM 2021.12.0
  * Help 1.0.3
  * parts_library
  * Reporting
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-01-31:
You need to recompute the wall. Closing this as a user error.

Please consider consulting the [forum](https://forum.freecad.org) before creating future GitHub issues. Thanks.

closed 2024-01-31 by Roy-043

nomennescio, 2024-01-31:
What does that have to do with Undo? To me this sounds like "Undo" is broken.

nomennescio, 2024-01-31:
> You need to recompute the wall. Closing this as a user error.
> 
> Please consider consulting the [forum](https://forum.freecad.org) before creating future GitHub issues. Thanks.

After the fact I found this article about bug submission guidelines:
https://forum.freecad.org/viewtopic.php?t=5236

I think that should be added to the Github bug reporting page if indeed that is the procedure that should be followed.

Roy-043, 2024-01-31:
On the GitHub issue page it says:
> 💬 START HERE -- FreeCAD Forum
You are encouraged to discuss the problem you are seeing on the FreeCAD Forums before opening an issue on GitHub