FreeCAD/FreeCAD #12222: [Problem] Applying an expression parameter to multiple selected objects only applies it to a single object

Anthony-Gaudino, 2024-02-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If I create two objects, for example two Part cubes and select both in the Model panel, and then change the Length property, it changes the property of both cubes.

If I do the same, but instead of changing the value directly I enter an expression, then it only applies the expression to one of the cubes.

I expect the expression to be applied to all selected cubes.

### Full version info

```shell
OS: macOS 14.3
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.9.22
  * ThreadProfile 1.89.0
  * Launcher
  * Nodes 0.1.36
  * InventorLoader 1.4.0
  * woodworking 0.21.2.33771
  * Render 2024.1.23
  * Silk 0.1.5
  * Defeaturing 1.2.2
  * nurbs
  * A2plus 0.4.62
  * Manipulator 1.5.7
  * fasteners 0.5.0
  * lattice2 1.0.0
  * CurvedShapes 1.0.5
  * sheetmetal 0.4.0
  * toSketch 1.0.1
  * Curves 0.6.22
  * Assembly4 0.50.7
  * Assembly3 0.12.2
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-03-06:
@Anthony-Gaudino can you attach a screencast of this ?

Anthony-Gaudino, 2024-03-07:
@luzpaz I hope this is what you meant:  https://github.com/FreeCAD/FreeCAD/assets/2439728/06a0d60b-7605-4a71-9b1c-1bef59426ce4

luzpaz, 2024-03-09:
Indeed! TY 
JFYI, you can just drag-drop those files into github textbox and it will automagically display: 
  
https://github.com/FreeCAD/FreeCAD/assets/4140247/e0a7d0b4-d805-4f0f-b67c-71b77c8c4f31

Anthony-Gaudino, 2024-07-12:
Updated Full version info as it still happens on 0.22.0dev.

furgo16, 2024-09-12:
I can confirm I can reproduce the issue as well.

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38695 (Git) Snap 1133
Build type: Release
Branch: main
Hash: 5785cb589bbb5f498f7f7de405f1fa521f9691d2
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Installed mods: 
  * OpenTheme 2024.9.1
  * Curves 0.6.46
  * ezydxf
  * Ondsel-Lens 2024.7.5.02
  * Assembly4 0.50.15
```

furgo16, 2024-09-12:
I'll add that it's not specific to Part Design. Easy to reproduce on the BIM workbench as well:

1. Create two walls.
2. Create a VarSet with e.g. a `wall_height` property. Set it to 2 m.
3. Select the two walls.
4. In the Properties editor, change the `Height` property to `<<VarSet>>.wall_height`
5. **Expected**: the height is changed to 2m for all walls
6. **Actual**: the height is changed to 2m only for one of the walls

furgo16, 2025-09-21:
Still relevant.

fivethreeo, 2026-03-16:
Still relevant

FEA-eng, 2026-10-25:
Still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 - Build 27.20.100.9664/Intel/Intel(R) UHD Graphics 620
```