FreeCAD/FreeCAD #12238: Techdraw view section misplace lines with links placement

michelsvd, 2024-02-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I created a FCStd file with two bodies, one with null placement not the other and another FCStd file with links to the first objects.
To see the two objects correctly in the 3D view, I have to duplicate the object placement on the link.
I create a view section in the second file. The position of the two objets in not respected. The difference is the placement value of the object with placement. May be the placement  of the link is added to the linked object placement in Techdraw, but the 3D view don't work that way.
The files are a cut down version of a complex project. And I cannot attach the FCStd files, even in a tgz.

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at b9bfa5c)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.7, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: French/France (fr_FR)
[/code]
```

### Subproject(s) affected?

Techdraw

### Anything else?

I put the FCStd in a tgz file but it says "We don’t support that file type", even if tgz is listed as supported!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-02-10:
You can attach an FCStd file by adding an acceptable extension to the end of the filename - "myFile.FCStd.zip" has worked for me in the past.

luzpaz, 2024-02-10:
@michelsvd please rename attachment or just use .zip

michelsvd, 2024-02-10:
Here they are
[fics.zip](https://github.com/FreeCAD/FreeCAD/files/14229978/fics.zip)

WandererFan, 2024-02-17:
This works correctly in v0.22dev.

closed 2024-02-17 by WandererFan