FreeCAD/FreeCAD #12239: Enhancement request - TechDraw add the ability to reference a user defined local directory for patterns or hashes.

RWL-69, 2024-02-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With the usage of appimages and snap packages, there is no set local install.  This can create problems for files that are referenced in to the directories in the appimages or snap packages.

Presently TechDraw has two options for various files, just as patterns or symbols.  It has been discussed in the forums for appimages (my interest).  

TechDraw has two paths for patterns, the appimage or snap image as well as a system wide local directory.  It would be nice to add a user controllable directory for TechDraw related files like hashes or patterns.  Presently only the Welding and Template directories can be directed to a user controlled directory.

In many cases, users do not have the ability to add files to the system wide directories for their usage.  This enhancement request will allow users to set up their own files in a local directory for use in TechDraw.

### Full version info

```shell
[code]
OS: Fedora Linux 39 (Thirty Nine) (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * DesignSPHysics 0.7.0(15-09-2023)
  * Assembly4 0.50.7
  * Plot 2022.4.17
  * Curves 0.6.23
  * symbols_library
  * ose-workbench-core-0.1.0a4
  * Defeaturing 1.2.2
  * ExplodedAssembly
  * FEM_FrontISTR 0.1.0
  * slic3r-tools
  * fasteners 0.5.0
  * OSE3dPrinter 0.1.0
  * InventorLoader 1.4.0
  * ose-workbench-core-0.1.0a4.zip
  * MeshRemodel 1.9.22
  * toSketch 1.0.1
  * ThreadProfile 1.89.0
  * freecad.gears 1.0.0
  * LCInterlocking
  * Pyramids-and-Polyhedrons
  * CurvedShapes 1.0.5
  * sheetmetal 0.4.0
  * Manipulator 1.5.7
  * kicadStepUpMod 10.23.0
[/code]
```

### Subproject(s) affected?

Techdraw

### Anything else?

[Bug 6706](https://github.com/FreeCAD/FreeCAD/issues/6707) which also refers to the discussion in the FreeCAD forum about the problems with appimages.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-05-18:
There is a preference for your favourite svg hatch file.  This can be set to a file in your local hatch file directory.  Once this is set, the file chooser for hatches will start in your local directory.

There is also a preference for the PAT hatch file.  That can also point to a local file/directory.

The only location missing is for Symbols, I think.

RWL-69, 2024-05-23:
Has this been implemented in the weekly builds?  I can test then.

luzpaz, 2024-08-09:
ping @WandererFan

benj5378, 2024-10-04:
@WandererFan, is this still relevant? We have to take a decision, whether we want to have separate preference entries for the paths themselves.

RWL-69, 2024-10-05:
I have not been working with TechDraw for awhile.

I feel it is still necessary.  If you use appimages, there is no defined directory between runs.  A random directory is created each time that FreeCAD is started.  Example, /tmp/.mount_FreeCAxdAAj9/usr/bin/freecad from one run and /tmp/.mount_FreeCAACwVxC/usr/bin/freecad this is the directory when FreeCAD restarted.  Using a recent weekly built appimage.  

The comment on May 18 about setting a default file does work to open a different file and works better than when I submitted my bug report.  

Under my quick tests today.  I think that for the 1.0 release, this issue can be closed as there is a workaround that seems to work in FreeCAD_weekly-builds-38827-conda-Linux-x86_64-py311.AppImage

luzpaz, 2024-10-14:
> Under my quick tests today. I think that for the 1.0 release, this issue can be closed as there is a workaround that seems to work in FreeCAD_weekly-builds-38827-conda-Linux-x86_64-py311.AppImage

what is the workaround ?