FreeCAD/FreeCAD #12240: [Problem] Chamfer001: Failed to create chamfer after modification

rmx-it, 2024-02-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Attached is a File with two bodies.
The first body is correct.
The second got build the same way, but after a modification the chamfer is failing.
The error is shown: Chamfer001: Failed to create chamfer after modification

[Ptrichter3.zip](https://github.com/FreeCAD/FreeCAD/files/14145031/Ptrichter3.zip)

The same chamfer is working as shown in body1
!?!

That seems to be a bug.
I`m using the latest weekly build on windows.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35759 (Git)
Build type: Release
Branch: main
Hash: 90d36f60c30cb6549a7987fcd9d4769eab1c160c
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.23
  * Help 1.0.3
  * lattice2 1.0.0
```
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-04:
@Roy-043 probably OCCT related and not currently not supported.
@rmx-it try a smaller fillet radius. e.g. 1,5 is working. Alternatively do a subtractive sweep with the 2mm on the edges.

Roy-043, 2024-02-04:
Most likely this is indeed OCC related. Note that the geometry in Sketch is different. It is not symmetrical relative to the X axis.

rmx-it, 2024-02-05:
> Most likely this is indeed OCC related. Note that the geometry in Sketch is different. It is not symmetrical relative to the X axis.

The sketch is symmetrical to the X axis.
The center of the circle segment is a 0 and both endpoints of the segments are aligned horinzontaly.

Roy-043, 2024-02-05:
The Sketch is almost, but not exactly symmetrical. The points where the circle connects to the lines have different X coordinates (the difference is 3.9mm).