FreeCAD/FreeCAD #12244: Bug in Techdraw of Freecad

ricou1, 2024-02-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The dimensions by Techdraw are not the same as in partdesignn. it is partdersign which is true

### Full version info

```shell
The dimensions by Techdraw are not the same as in partdesignn. it is partdersign which is true
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-02-03:
Please re-edit report. Add your full [About](https://wiki.freecad.org/About) info in the **Full version info**  portion. Add screenshots of what the problem is, zip up an example file and drag-drop into github here as an attachment. TIA

ricou1, 2024-02-03:
Sorry i was not squcessfull to paste an picture, in a html page; so i send
you an email.
There is  a problem in the dimensions given par tyechdraw. See the two
picture of the same3 object. After verification <with a caliper techdraw as
wrong, partdde3s36ign3 is good.
si30n3ce3re3ly 3yours.

luzpaz, 2024-02-03:
Emails don't work with attachments. 
Go back to github, zip the freecad file (in to a .zip) and then drag and drop into github text box (the area where you can type replies intoi. 

For attaching images: right click on a picture and copy then ctrl+v or drag-drop into github text box.

Roy-043, 2024-02-04:
@ricou1 You have now created a 3rd issue for the same problem! That is not the right way to do things. Please do not do this again.

Closing this as a duplicate.

closed 2024-02-04 by Roy-043

WandererFan, 2024-02-10:
If you select geometry on the TechDraw page, you will get the dimension of the projected geometry and the "Measure Type" will be "Projected".  This will only be accurate if the projection direction is orthogonal to the dimension line.  If you are dimensioning an axonometric view, your dimension value will be incorrect.

If you want to use the actual dimension from the 3d model, then you should select the geometry in the 3d window.   This will create a "True" dimension.

You can use "repair references" to change from projected geometry to actual geometry.

bernerus, 2024-02-11:
Maybe this is another issue, but at least related. 
When I try setting "Measure Type" to true, on an axonometric view, the value 0 (zero) is shown. Repairing references does not help.

ricou1, 2024-02-11:
Yes that is that I have seen

WandererFan, 2024-02-11:
You can only have a true measurement if you have 3d references.  

If you are using 2d references and set the Measure Type to true, there should be an error message generated.

bernerus, 2024-02-12:
Agree, setting the value to 0 is just confusing and leaves no hint on what's wrong.

-- Chris