FreeCAD/FreeCAD #12253: [Problem] Very simple sketch causes crash when moving any object

jhgarrison, 2024-02-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Total beginner here. I created a very simple sketch (house shape).  At one point I attempted to constrain a line to lie on the x-axis, and the program hung ("not responding").  After restarting, I can open the sketch, but any attempt to move anything causes the same hang.  The file in question is attached.

[play1.zip](https://github.com/FreeCAD/FreeCAD/files/14154216/play1.zip)

1. Open the file
2. Double-click the sketch in the model view
3. Attempt to drag any component

The dragged object moves slightly and then the program will immediately hang.

I launched FreeCAD with the `-l` option and captured a log file:

[FreeCAD.log](https://github.com/FreeCAD/FreeCAD/files/14154252/FreeCAD.log)

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jhgarrison, 2024-02-04:
Maybe it fails on Windows 11?

This is literally the very first sketch I created after installing FreeCAD with the Windows installer.

I tried the latest weekly build, and it fails the same way.  Looking at the system state with ProcessExplorer I see that FreeCAD is consuming about 6.5% CPU and zero I/O deltas which means, on my 16-core system, that it's stuck in a 100% CPU loop.

I could use some suggestions on how to provide more debug information.  (My software background is primarily Java)

jhgarrison, 2024-02-04:
I tried a few other operations, like simply creating a line.  These also freeze immediately after I click the mouse in the sketch after selecting a tool.  I made a screen capture video but don't see the "attach file" option in the GitHub UI for adding comments (and "insert image" doesn't like .mp4), so I put a copy on Google Drive [here](https://drive.google.com/file/d/16jeD1xB48TVm3yN7EFZ1KHm6IU1JtfXk/view?usp=sharing).  

(Do you need to enable "attach file" for this project, or is "attach file" only available when creating an issue?)

jhgarrison, 2024-02-04:
In case it's useful...

[image]

Roy-043, 2024-02-07:
Maybe it is related to the graphics card or driver.
Try switching to "Use software OpenGL":
https://wiki.freecad.org/Preferences_Editor#3D_View

xtemp09, 2024-02-09:
@jhgarrison, you may pinpoint the code fragment if you launch a debug version of FreeCAD in a debugger, like _cdb_, _lldb_ or _gdb_. If this is impossible, running a release version in a debugger could help as well.

luzpaz, 2024-03-06:
@jhgarrison :point_up_2:

maxwxyz, 2024-06-13:
@jhgarrison can you provide the info?

maxwxyz, 2024-06-13:
I'll close this then

closed 2024-06-13 by maxwxyz