FreeCAD/FreeCAD #12254: Problem of dimensions in tech draw

ricou1, 2024-02-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello,
I think that there is a problem in techdraw dimensions. After verification with a caliper tech draw are wrong, but part design are good. 
[verif_techdraw.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/14155562/verif_techdraw.FCStd.zip)

### Full version info

```shell
Hello,
I think that there is a problem in techdraw dimensions. After verification with a caliper tech draw are wrong, but part design are good.
```

### Subproject(s) affected?

Techdraw

### Anything else?

verif_techdraw.FCStd.zip

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-02-04:
Your TechDraw dimensions are PROJECTED dimensions. You need to get better acquainted with TechDraw and also read the manual: https://wiki.freecad.org/TechDraw_LengthDimension#Display_3D_measurement

You are very much a beginner. Please consult the forum first before creating future GitHub issue.

Closing this as a user error.

closed 2024-02-04 by Roy-043

ricou1, 2024-02-04:
Thank you because techdraw is very useful