FreeCAD/FreeCAD #12255: [Problem] no focus return to `Model` after `Part Design/Tasks/Close`

Andrei-Pozolotin, 2024-02-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When it works:
* open `Sketch`, focus is on `Tasks`
* close `Sketch`, focus is on `Model` **<---OK**

When it fails:
* open `Pad`, focus is on `Tasks`
* close `Pad`, focus is on `Tasks` **<--- PROBLEM**

Attached:
* sample screen sequence
* sample freecad document

### Full version info

```shell
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35806 (Git)
Build type: None
Branch: makepkg
Hash: 1bb5350c5102362889be02c14304eaefb5bee6c0
Python 3.11.6, Qt 5.15.12, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
```
```

### Subproject(s) affected?

None

### Anything else?

<details>
<summary>
1-initial-state
</summary>

![1-initial-state](https://github.com/FreeCAD/FreeCAD/assets/1622151/b488aa39-0742-4e5f-b61c-f75a57f5e3f1)

</details>

<details>
<summary>
2-sketch-open
</summary>

![2-sketch-open](https://github.com/FreeCAD/FreeCAD/assets/1622151/582fe98e-b8bb-4ad1-8448-16dcff938eb5)

</details>

<details>
<summary>
3-sketch-close
</summary>

![3-sketch-close](https://github.com/FreeCAD/FreeCAD/assets/1622151/f536e1cf-0c6e-4bfa-b502-429bb059f628)

</details>

<details>
<summary>
4-design-open
</summary>

![4-design-open](https://github.com/FreeCAD/FreeCAD/assets/1622151/cbebf473-b533-4aa9-9b02-3e3ef58851c1)

</details>

<details>
<summary>
5-design-close
</summary>

![5-design-close](https://github.com/FreeCAD/FreeCAD/assets/1622151/97e376a6-1b19-4e97-963a-ef6d698defbe)

</details>

<details>
<summary>
sample file
</summary>

[issue-focus.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/14155845/issue-focus.FCStd.zip)

</details>

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Andrei-Pozolotin, 2024-02-04:
### Subproject(s) affected?

None

### Anything else?

<details>
<summary>
1-initial-state
</summary>

[image]

</details>

<details>
<summary>
2-sketch-open
</summary>

[image]

</details>

<details>
<summary>
3-sketch-close
</summary>

[image]

</details>

<details>
<summary>
4-design-open
</summary>

[image]

</details>

<details>
<summary>
5-design-close
</summary>

[image]

</details>

<details>
<summary>
sample file
</summary>

[issue-focus.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/14155845/issue-focus.FCStd.zip)

</details>

benj5378, 2024-10-04:
I cannot reproduce this. Has it been fixed? Can you please try again on newest version, @Andrei-Pozolotin?

If no news within 14 days, then close this

luzpaz, 2024-10-16:
@Andrei-Pozolotin :point_up_2:

Andrei-Pozolotin, 2024-12-14:
yes, the problem is still present as of:
```
OS: Arch Linux (KDE/plasmax11/xcb)
Architecture: x86_64
Version: 1.1.0dev.39622 (Git)
Build type: Release
Branch: makepkg
Hash: f5a4ec01f8b095051355226be735104ce8eb8b44
Python 3.12.7, Qt 6.8.1, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/breeze
```

luzpaz, 2025-06-11:
@Andrei-Pozolotin can you produce on latest 1.1dev (qt6) ?