FreeCAD/FreeCAD #12258: [Problem] Unable to fillet symmetrical side

YosypC, 2024-02-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When created individually, it works. With both, it gives me an error. Before I was able to fillet both without a significant model update (the vertex at the bottom had 4 edges just like this case). By selecting all sides, it completes fine. Even by selecting the lowest edge that connects everything, it still yields an error.

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35315 (Git) AppImage
Build type: Release
Branch: main
Hash: 09f67f2f6bff1c57ac8019a5ba2f4a2c41025ee6
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * OpenDark 2023.12.17
```

### Subproject(s) affected?

PartDesign

### Anything else?

[image]
[image]
[image]
[image]
[image]
[Desk (copy).zip](https://github.com/FreeCAD/FreeCAD/files/14157133/Desk.copy.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-17:
Will close this as you've mentioned. This is also depending on OCC.

closed 2024-03-17 by maxwxyz

YosypC, 2024-03-17:
Alright. Maybe attaching OCC tag would be useful?