FreeCAD/FreeCAD #12259: [Problem] no way to discover `ActiveObject` from `python`

Andrei-Pozolotin, 2024-02-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

PROBLEM:

currently, for an external macro/plugin there is no way to discover an `ActiveObject`:
* since the `Document.ActiveObject` is unusable, see
#11754
* and the `Selection.getSelection()` is broken, see below

SOLUTION:

perhaps it would be easier to to fix `Document.ActiveObject`, 

since `Selection.getSelection()` is currently overloaded with two different functions: 
* to reflect actual logical object selection for the code
* and to show "clean selection representation" for the user

for example, in `src/Mod/PartDesign`:
https://github.com/FreeCAD/FreeCAD/blob/c5d0fbd7071f4b89f820c285c58cb542f7c7c0b3/src/Mod/PartDesign/Gui/Command.cpp#L569

VERIFICATION:

for `App.Document.ActiveObject`, `Gui.Document.ActiveObject`, see #11754

for `Selection.getSelection()`, when it works:
* create, close, open a `Spreadsheet`
* execute and observe in `python console`: selection is **OK**
```python
Gui.Selection.getSelection()
[<Sheet object>]
```

for `Selection.getSelection()`, when it fails:
* create, close, open a `Pad`
* execute and observe in `python console`: selection is **MISSING**
```python
Gui.Selection.getSelection()
[]
```

### Full version info

```shell
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35815 (Git)
Build type: None
Branch: makepkg
Hash: c5d0fbd7071f4b89f820c285c58cb542f7c7c0b3
Python 3.11.6, Qt 5.15.12, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-03-03:
I guess `Document.ActiveObject` is simply not what you expect. It's just a pointer to the last added object of a document and it has no relation to the selection or what's shown in the tree view.

luzpaz, 2024-03-06:
@wwmayer how would you approach this ?

wwmayer, 2024-03-06:
Because the expectation is wrong the whole bug report is wrong.

closed 2024-08-27 by luzpaz