FreeCAD/FreeCAD #12260: [Problem] Start page - Blog has no Tabs to go back

Palmstroemen, 2024-02-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The nice Start page inside FreeCAD has 4 tabs. "Documents", "Help", "Activity", "Blog".
You would expect, that when selecting a of these tabs you can always go back and select another tab. This works perfectly for the first 3 tabs.
[image]
 But when you click on "Blog" all of a sudden your tabs are gone and there's no way back.  

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * A2plus 0.4.62
  * AnimationFreeCAD 1.0.0
  * Assembly3 0.12.2
  * Assembly4 0.50.6
  * boltsfc 2022.11.5
  * btl 0.9.9
  * CADExchanger
  * CfdOF 1.25.1
  * DesignSPHysics 0.7.0(15-09-2023)
  * dodo 1.0.0
  * DynamicData 2.60.0
  * Estimate 0.1.2
  * ExplodedAssembly
  * ExtremeProDark 2.7.0
  * fasteners 0.5.0
  * frame 0.1.1
  * FreeCAD-addons
  * freecad.gears 1.2.0
  * Glass
  * IconThemes
  * InventorLoader 1.4.0
  * lattice2 1.0.0
  * MakerWorkbench 1.0.1
  * Manipulator 1.5.7
  * Nodes 0.1.36
  * PieMenu 1.2.7
  * pluginloader
  * pluginloader.backup1702937756.6938357
  * POV-Ray-Rendering
  * RemBench
  * sheetmetal 0.4.0
  * ShortCuts
  * ShortCutsDeveloper
  * Silk 0.1.5
  * WB_Organizer 2024.1.29
  * WebTools 1.0.0
  * workfeature
  * workfeature-macro
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-02-04:
You can go back with the option from the right-click context menu. Or use the Web_BrowserBack command from the Navigation toolbar. In both cases you have to refresh the page then.

Roy-043, 2024-02-04:
Maybe this is not that strange? It is after all a web-browser.

Palmstroemen, 2024-02-04:
But would it be difficult to add the tabs anyway?

Roy-043, 2024-02-04:
The Blog is an external website. The other pages are generated in FreeCAD. Adding a tab to the Blog would not make sense. Maybe an external-link arrow can be added to the Blog tab (which has the appearance of a button in V0.22 BTW)?

Palmstroemen, 2024-02-05:
But why not show the external browser inside the Tab-Widget in a QtWidget?
Right now it's also sitting somewhere inside the Gui-Widgettree. So why not set it into the widget that shows the other 3 pages?

Palmstroemen, 2024-02-05:
No problem. The webbrowser stays the same INSIDE the tabbed FreeCAD QtWidget. Now it seems to sit somehow inside the FreeCAD mainwindow. In case you already have a part or a spreadsheet open, you have multiple windows with tabs on the bottom. "Start", "loadedFile". If you go into the Blog you still have these two tabs. So I assume, the two pages (Browser, 3Dview) sit in a QtWidget which is their parent. 

Now inside the "Start" page, there is another tabbar. Maybe a child of the QtWidget around it. This tabbar has 4 subpages, but the 4th (Blog) seems not to sit in a Subwidget of this TabBar but somehow outside.

yorikvanhavre, 2024-02-06:
We could display the blog inside an iframe inside the "frame" of the start page. But we are actually starting to have a serious discussion about removing the web browser structure of FreeCAD, because it begins to give serious problems. That means the start page, which depends on it, would need to be rewritten and use Qt widgets instead.

So it probably does not make much sense to spend much work on the current version of the start page... Also because by hitting the "back" button of the start WB you get back to the startpage with the 4 tabs.

So I think it's best to close this issue as "won't do", but feel free to reopen if you feel this is a wrong move.

closed 2024-02-06 by yorikvanhavre