FreeCAD/FreeCAD #12265: [Problem] V0.22 Sketcher Elements panel is greyed out and therefore seems disabled

Roy-043, 2024-02-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the version listed below the icons in the Elements panel of the Sketcher dialog appear greyed out. This sends the (strong) signal to the user that these buttons are disabled. In addition the icons are coarsely pixelated and seem to be scaled unequally. The fact that they are almost monochrome is also problematic since colors are used to distinguish them.
[image]

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35803 (Git)
Build type: Release
Branch: main
Hash: 90c90f705bec20faf4b8bb419574134c33c7d4b5
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Sketcher

### Anything else?

Forum topic:
https://forum.freecad.org/viewtopic.php?t=84909

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-02-07:
Qt can generate images. So maybe it is possible to add the background to the image? Or, alternatively, add a border to the image (transparent/some-qss-contrast-color).

kadet1090, 2024-02-07:
This is not a problem with generating images (this can be done with QPainter) but with making it themeable with qss and calculating the proper size for background that includes borders, paddings etc. Unfortunately Qt does not expose QSS data anywhere.

I know the rect of icon, so I could in theory paint 2px bigger background under it - but I don't know how to obtain proper color which was an issue for OpenDark where item background and background painted under icons were the same as the items background. It would also not be possible to control any spacing, border etc via qss. This however my by my skill issue as I simply don't have Qt experience. I'll try to come up with better solution but hard to promise anything right now.

Roy-043, 2024-02-08:
I think you have misunderstood my proposal. My suggetion is to avoid the theming issues by creating an image with an alternate background or a border at runtime. If a sensible background or border color cannot be retrieved, you can add a double border (white and black) to the image. This border would be invisble (transparent) for unselected icons.

FEA-eng, 2024-06-26:
@PaddleStroke or @kadet1090  Could you have a look at this? It's quite confusing and the scaling issue makes it particularly conspicuous.

kadet1090, 2024-06-26:
@maxwxyz can you assign me? We need at least to fix scaling issues.

maxwxyz, 2024-08-14:
The icons look squashed too, definitely a regression from the stable version:
[image]

Roy-043, 2024-09-01:
Wouldn't it be better to revert this to the V0.21 solution?

kadet1090, 2024-09-01:
this would introduce another problems with checkboxes, PR with this fixed few other issues

kadet1090, 2024-09-01:
[image]

This is how it looks for me, I cannot reproduce this issue hence it is hard to fix it. We need more information with such reports such as theme used, OS used, hidpi settings, Qt etc as it all can impact how it works. Also it seems that this issue should really be two separate issues:

1. Squashed and wrongly scaled icons - which is clearly a bug needs to be addressed somehow
2. Semi-transparent non-selected geometry - which is not a bug, but there maybe a better solution.
 
As the icons are only semi transparent and non grayed out it should be quite easy to learn what it means. This needs further discussion as previous solution also was not really good, I only understood what was presented on screen after reading code

Roy-043, 2024-09-01:
This may only affect Windows user. But a significant part of our user base use that OS. From my side it looks as if a minor issue has been solved at the expense of introducing a mayor one.

kadet1090, 2024-09-01:
Which one? The icon scaling issue will have to be addressed somehow, I'm thinking how to pinpoint it more accurately - can you test if this is still a problem while using Fusion as QStyle? And do you experience a problem with rendering checkboxes?

PR #16265 (2024-09-04): Gui: Fix issues with icons in sketcher elements panel
This should fix issues described in #12265. 

For strange icon scaling I forced 24px icon size - this should ensure that icons have proper sizing. I however could not reproduce the issue in any way on my setup so it is blind fix. We need someone to test if it does indeed work.

For the issue with dimmed out icons I made them opaque unless the (pre)selection is partial. That way normally icons are fully opaque and only when hovered or something is selected they become semi-transparent.

Before:
[image]

After:
[image]

Fixes: #12265
closed 2024-09-04 by chennes