FreeCAD/FreeCAD #12266: [Problem] FreeCAD switches document right after entering a Sketch

YosypC, 2024-02-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The active document get switched after entering in a main document's sketch.

Related to realthunder/FreeCAD#792

https://github.com/FreeCAD/FreeCAD/assets/50027752/4f03a851-210d-48b7-bdf7-73a2a3ef51bd

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35315 (Git) AppImage
Build type: Release
Branch: main
Hash: 09f67f2f6bff1c57ac8019a5ba2f4a2c41025ee6
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * OpenDark 2023.12.17
```

### Subproject(s) affected?

None

### Anything else?

```
Python 3.10.13 | packaged by conda-forge | (main, Oct 26 2023, 18:20:51) [GCC 12.3.0] on linux
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> # Gui.Selection.clearSelection()
>>> # ActiveSketch = App.getDocument('Unnamed').getObject('Sketch')
>>> # tv = Show.TempoVis(App.ActiveDocument, tag= ActiveSketch.ViewObject.TypeId)
>>> # ActiveSketch.ViewObject.TempoVis = tv
>>> # if ActiveSketch.ViewObject.EditingWorkbench:
>>> #   tv.activateWorkbench(ActiveSketch.ViewObject.EditingWorkbench)
>>> # if ActiveSketch.ViewObject.HideDependent:
>>> #   tv.hide(tv.get_all_dependent(App.getDocument('Unnamed').getObject('Body'), 'Sketch.'))
>>> # if ActiveSketch.ViewObject.ShowSupport:
>>> #   tv.show([ref[0] for ref in ActiveSketch.Support if not ref[0].isDerivedFrom("PartDesign::Plane")])
>>> # if ActiveSketch.ViewObject.ShowLinks:
>>> #   tv.show([ref[0] for ref in ActiveSketch.ExternalGeometry])
>>> # tv.sketchClipPlane(ActiveSketch, ActiveSketch.ViewObject.SectionView)
>>> # tv.hide(ActiveSketch)
>>> # del(tv)
>>> # del(ActiveSketch)
>>> # 
>>> import PartDesignGui
>>> # ActiveSketch = App.getDocument('Unnamed').getObject('Sketch')
>>> # if ActiveSketch.ViewObject.RestoreCamera:
>>> #   ActiveSketch.ViewObject.TempoVis.saveCamera()
>>> #   if ActiveSketch.ViewObject.ForceOrtho:
>>> #     ActiveSketch.ViewObject.Document.ActiveView.setCameraType('Orthographic')
>>> # 
>>> # App.setActiveDocument("Desk")
>>> # App.ActiveDocument=App.getDocument("Desk")
>>> # Gui.ActiveDocument=Gui.getDocument("Desk")
>>> # Gui.getDocument('Unnamed').resetEdit()
>>> App.ActiveDocument.recompute()
>>> # ActiveSketch = App.getDocument('Unnamed').getObject('Sketch')
>>> # tv = ActiveSketch.ViewObject.TempoVis
>>> # if tv:
>>> #   tv.restore()
>>> # ActiveSketch.ViewObject.TempoVis = None
>>> # del(tv)
>>> # del(ActiveSketch)
>>> # 
>>> # Gui.Selection.addSelection('Unnamed','Body','Sketch.')
>>> App.getDocument('Unnamed').recompute()
>>> # App.setActiveDocument("Unnamed")
>>> # App.ActiveDocument=App.getDocument("Unnamed")
>>> # Gui.ActiveDocument=Gui.getDocument("Unnamed")
>>> # App.setActiveDocument("Desk")
>>> # App.ActiveDocument=App.getDocument("Desk")
>>> # Gui.ActiveDocument=Gui.getDocument("Desk")
>>> # Gui.Selection.clearSelection()
>>> 
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-02-08:
I have this issue too! It also switches to start page. 
Seems to be gone when properties panel is open, which I happily closed with the new right click option. @PaddleStroke ?

luzpaz, 2024-08-29:
Is this still occurring ?

maxwxyz, 2024-09-01:
The greedy focus of Start page should've been fixed iirc
@YosypC @MisterMakerNL could you confirm on the latest weekly?

YosypC, 2024-09-01:
@maxwxyz Can't. I'm out for work and I don't even have it installed on my current computer :(

MisterMakerNL, 2024-09-01:
I have not noticed this issue for a while, also I tried to reproduce and failed. So for from my side it can be closed/

maxwxyz, 2024-09-01:
I'll close it then. If it occurs, please reopen.

closed 2024-09-01 by maxwxyz