FreeCAD/FreeCAD #12267: [Problem] Wrong created external geometry

YosypC, 2024-02-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]

I selected the blue line but it projected the remaining, non drawn part of the circumference.

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35315 (Git) AppImage
Build type: Release
Branch: main
Hash: 09f67f2f6bff1c57ac8019a5ba2f4a2c41025ee6
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * OpenDark 2023.12.17
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-11:
can reproduce. @PaddleStroke maybe this can be tackled together with https://github.com/FreeCAD/FreeCAD/issues/11824

drew585, 2024-07-31:
I have what seems like the same issue.
I'm sketching on XZ plane. I selected the left side radius. The line created does not have a point on the far left of the part.
Looks like its shifted to the right of where it should be.
[image]
Problem is on FC 0.22 38334.
Linux mint 22 64 bit.

I tried it in FC 0.21.2 and it works correctly there.

maxwxyz, 2024-08-01:
@bgbsww @PaddleStroke @AjinkyaDahale any idea how this regression came into dev?

drew585, 2024-08-03:
I just tested this in the new build 38385 and it seems to be working correctly now.

maxwxyz, 2024-08-03:
@YosypC can you confirm with today's weekly?

drew585, 2024-08-03:
I just now tested it with new sketches on the same model in both software versions.

38334
[image]

38385
[image]

THANK YOU!!!!

closed 2024-08-03 by maxwxyz