FreeCAD/FreeCAD #12271: [Problem] cannot print in A4 landscape using techdraw

albmef, 2024-02-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

hi all
cannot print in A4 landscape using techdraw
steps to reproduce:
-techdraw -> add page using blanck A4 landscape model
-add a view
-file -> print preview -> preview is OK
-print from preview dialog -> output is PORTRAIT and CROPPED
-try changing landscape to portrait in printer options -> output is half size, like A5 landscape in A4 portrait.

sorry maybe i did wrong but i found no settings about printing in edit->prefs...

thanks for the support
best regards
AM

### Full version info

```shell
[code]
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) Snap 908
Build type: Release
Branch: tag: 0.21.2
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.7, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: French/France (fr_FR)
[/code]
```

### Subproject(s) affected?

Techdraw

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-02-05:
TechDraw is designed work work as a PDF export and then be printed as per https://wiki.freecad.org/Std_PrintPdf

closed 2024-02-05 by albmef

WandererFan, 2024-02-10:
You should not need to export to pdf before printing.  File > Print should be enough.

If you print directly (without going through print preview), does it work better?  It seems to me that I've had trouble printing from the preview.

albmef, 2024-02-10:
hi @WandererFan 
i reproduced what you said using File->Print directly without preview. 
same thing happened unfortunately. 
howto :
- techdraw based on A4 landscape blank model
* first attempt
- file->print
- printer properties -> landscape
- start printing : output in portrait cropped
* second attempt
- file->print
- printer properties -> portrait
- start printing : warning popup (printed has a different orientation than the drawing), the output is portrait but only the upper half of the sheet is printed, and the drawing is scaled down to fill the page width

my printer is an A4 BW laser printer, a Brother MFC-L2500DW, which works perfectly with ;S Zindozs, MACOS and Ubuntu.
thanks for your support
best regards
AM

WandererFan, 2024-02-11:
This looks a lot like what happens when I send an A4 print to my printer (which is expecting US Letter size). 

Just to narrow this down a bit, if you print to Pdf, does it come out correctly?

albmef, 2024-02-11:
sure i did ! both techdraw page pdf export from freecad and printing from pdf (using Okular) worked as expected. Landscape orientation was kept all the way through.
AM

HiFiJ, 2025-06-08:
Hi everyone, thanks for the stupendous work on FreeCAD, absolutely loving it

I believe I am facing the same issue as the OP. I am trying to print a landscape Page (SVG A4 Landscape Blank Template) on a current ArchLinux with the Edge Snap (details below), but no matter what it prints as a cropped portrait.

Exporting to PDF is working corretly regarding Landscape layout; printing from an external PDf viewer then gives a correct landscape print.

Thanks & best regards,

Joost

OS: Ubuntu Core 22 (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.0dev.42099 (Git) Snap 1485
Build date: 2025/06/08 00:31:19
Build type: Release
Branch: main
Hash: 8a58d225c822d20fcf973a8de8b2cd5095d7b31e
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, boost 1_74, Eigen3 3.4.0, PySide 5.15.11
shiboken 5.15.11, SMESH 7.7.1.0, xerces-c 3.2.3, IfcOpenShell 0.8.2, OCC 7.7.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/212.454/1

WandererFan, 2025-06-09:
works here:

[image]

on this:

OS: Linux Mint 21.3 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.41904 +6 (Git)
Build date: 2025/06/08 14:33:35
Build type: Unknown
Branch: DetailIssues
Hash: c82f0b6365049961d292e0c08ad878463bd85fdf
Python 3.10.12, Qt 5.15.2, Coin 4.0.0, Vtk 7.1.1, boost 1_74, Eigen3 3.4.0, PySide 5.15.2
shiboken 5.15.2, SMESH 7.7.1.0, xerces-c 3.2.3, OCC 7.8.0
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 115/141.585/1
Installed mods: 
  * fasteners 0.5.21