FreeCAD/FreeCAD #12278: Body placement not respected when exporting to STL

Seahorde, 2024-02-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a model with parts, bodies and features. Some parts and bodies have placement set: axis, angle and position. The model displays perfect when in freecad application. Once I export it to STL, I see that the body placements aren't respected and bodies that have position set are rendered in wrong places.

When I set the body placement to zeroes and transfer offsets to the sketch placements inside the body, everything looks OK in the exported STL.

### Full version info

```shell
0.21.2 rev 33771
```

### Subproject(s) affected?

File formats

### Anything else?

Here's an example. The "Connector" body is affected:
[Oil Funnel.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/14190018/Oil.Funnel.FCStd.zip)

Looks OK in FreeCAD app:
[image]

STL is broken:
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Seahorde, 2024-02-07:
In the example above the placement angle of the bodies isn't respected as well.

Syres916, 2024-02-08:
@Seahorde while I agree this appears to be a legitimate bug and should be investigated as such, you have modeled a single contiguous solid (judging by the FreeCAD screenshot) and yet you have used more than one Body and even used Part containers, all of which is unnecessary IMHO.

Part containers are where two or more non-contiguous solids are being 'assembled' so for example a plate with four tapped holes and the four fasteners that must be aligned with the plate at all times would be a perfectly acceptable use.

If you are struggling to understand how to model this as one contiguous solid then perhaps the [forum ](https://forum.freecad.org/viewforum.php?f=3) might be a good place to ask, there's plenty of PartDesign power users who will be able to advise if it's possible.

MisterMakerNL, 2024-02-09:
Works fine in 0.22 when you move the parts out of the part:
[image]
Please don't put parts in your parts so you can do parts in your parts.
Freecad is confusing enough without this :)

Seahorde, 2024-02-10:
I put different components into separate Bodies because:
a) Having a complex body with 10x features isn't practical as the tip easily rips apart when I'm modifying some internal features. Not to mention filets and chamfers - they are almost 100% times ruined if the source sketch and features get changed.
b) I want to shift/rotate a group of features and paint them different color. Putting them into an extra Body works the best.

BTW, I needed to re-export a model I've worked on a year ago and exported into 3mf. Now I need to export it to STL. I opened it with the recent FreeCAD, exported and immediately saw that the bug I'm talking about here is a regression. It wasn't there a year ago.

FreeCAD 0.21.2:
[image]

STL:
[image]

Seahorde, 2024-02-10:
A workaround: Create a Part wrapper around all bodies and transfer the body positioning to the wrapper part. It's ugly, but it at least works.

[image]

maxwxyz, 2024-02-11:
@Seahorde is it working in the current dev versions for you?

Syres916, 2024-02-11:
@maxwxyz the whole issue of implications of Part containers has been going on for a number of years and as MisterMakerNL comment https://github.com/FreeCAD/FreeCAD/issues/12278#issuecomment-1936596523 using 0.22 is still not fixed. I'd love to go back and from version 0.16 see when the misbehaviour started but that's quite a bit of work in it's own right testing all the different scenarios, not just this one. Someone reported that 11 tree depth is OK but as soon as they went to 12 deep really weird things start to play out but again I cannot prove that yet and when it started.

Even importing a known good Step assembly which has all the part containers pre-populated results in odd components looking like they've tried to escape, 99% are properly assembled!! I just never use Part containers as I cannot trust what behaviour they are going to create.

boucaron, 2024-03-16:
In my case, I used the following workaround : export as OBJ, reimport the OBJ and finally export as STL. 

I had the issue on Extruded sketches. On those sketches I performed a transformation : in one case on the position, and the other case it was both the position and the rotation.

SergiiChernov, 2024-03-24:
> In my case, I used the following workaround : export as OBJ, reimport the OBJ and finally export as STL.
> 
> I had the issue on Extruded sketches. On those sketches I performed a transformation : in one case on the position, and the other case it was both the position and the rotation.

In my case *(0.21.2) obj fails as miserably as stl but step seems to work... so far.
Absolutely awful and unacceptable bug that stays unfixed for years.

sylvainbarriere, 2024-05-23:
Same issue really annoying , seems not present in version 0.21.1 so I open 0.21.1 instance to export my part

languagelawyer, 2024-06-01:
Happens to me in 0.21.2 (b9bfa5c5507506e4515816414cd27f4851d00489), doesn't in a ±fresh daily build (0.22.0dev 7d6e9b19fb6a5a5d4f29ef8c95ff43848648d9dd)

maxwxyz, 2024-06-01:
@Seahorde could you confirm that this is working in the latest weekly from: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Seahorde, 2024-06-02:
@maxwxyz I tried a simple test with two displaced bodies and indeed, the STL export looks correct now! Wow!
I will do some more modeling soon and let you know if I find any issues.

CamNorm, 2024-06-08:
Same trouble with 3d objects create with the partdesign tool:
Exported with the 0.19 Freecad version works perfect, but the same Fcad file export from 0.21.2 bugs

My model in FreeCAD
[image]
The exported STL with 0.19
[image]
The exported STL with 0.21.2
[image]

luzpaz, 2024-07-31:
Per the discussion, this issue is fixed in the latest development 0.22.x version. A fix won't be backported to 0.21.x

winstonford, 2024-11-15:
Having a similar issue in 1.0rc4 when exporting to DXF.  Body placement not respected when exporting to DXF.  However, there is no issue when exporting same file in latest stable 0.21.2

I create a new file in 1.0rc4 and in sketcher create a sketch with a centered fully constrained square.  I create another sketch with a centered fully constrained circle inside the square and concentric with it.

With the circle sketch selected, In the data pane under Position, I set X to move circle outside square.

I select both sketches and export to dxf.  I open the dxf in librecad, Inkscape, and sharecad.  The circle is back inside the square.  It should be outside the square, but the position is not respected.

I open the file in 0.21.2 and repeat export and open steps.  There is no issue, the circle is outside the box.

The file and the 2 exported dxf files are attached.

[outside-the-box.zip](https://github.com/user-attachments/files/17760371/outside-the-box.zip)

luzpaz, 2024-11-15:
@winstonford full [About](https://wiki.freecad.org/About) info please

maxwxyz, 2024-11-16:
I cannot reproduce the original issue in the latest weekly
```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39176 (Git) Conda
Build type: Release
Branch: main
Hash: 45d3864942c6928cec7dc18f95b1f222485756e5
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * CfdOF 1.27.14
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.5.3
  * Silk 0.1.5

```

The new comment is about DXF export, not STL - not sure if reopening this STL issue is the best idea. @winstonford could you please open a new issue for this? Maybe there is a similar already reported: https://github.com/FreeCAD/FreeCAD/issues?q=is%3Aissue+is%3Aopen+dxf+export+label%3A%22File+format%3A+DXF%22

closed 2024-11-16 by maxwxyz