FreeCAD/FreeCAD #12279: A problem caused this program to stop interacting with Windows.

CharlieK1968, 2024-02-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I  am making a holder for 18650 standing batteries, when adding in the holes for the batteries, it crashed every time. It does not like something I am doing and just goes unrespeisive.

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60n
  * Curves 0.6.13
[/code]
```

### Subproject(s) affected?

None

### Anything else?

Description
A problem caused this program to stop interacting with Windows.
Faulting Application Path:	C:\Program Files\FreeCAD 0.20\bin\FreeCAD.exe

Problem signature
Problem Event Name:	AppHangB1
Application Name:	FreeCAD.exe
Application Version:	0.20.2.2
Application Timestamp:	640cf95b
Hang Signature:	2204
Hang Type:	134217728
OS Version:	10.0.22621.2.0.0.768.101
Locale ID:	1033
Additional Hang Signature 1:	22047e28f797c64e172430f3014355b7
Additional Hang Signature 2:	f331
Additional Hang Signature 3:	f3312ddf0d798976e99182a0dc24d706
Additional Hang Signature 4:	2204
Additional Hang Signature 5:	22047e28f797c64e172430f3014355b7
Additional Hang Signature 6:	f331
Additional Hang Signature 7:	f3312ddf0d798976e99182a0dc24d706

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-02-07:
Without a file and step-by-step instructions on how to reproduce there nothing we can do with this issue. Also note that you are not using the current stable version (V0.21.2).

Roy-043, 2024-02-10:
No useful information has been provided. Closing this one.

closed 2024-02-10 by Roy-043