FreeCAD/FreeCAD #12287: [Problem] Items are moved in reverse order, breaking dependencies

toddmo, 2024-02-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When selecting multiple steps in the model, and then choosing "Move after..." from the context menu, the items are moved but their order flips from source to destination, making them in reverse dependency order at the new location. This makes no sense. It breaks the items in question and n more move operations have to be done to fix it, with each move involving a lengthy wait for re-computation.

### Full version info

```shell
[code]
OS: Linux Mint 21.2 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * ProDarkThemePreferencePack 1.0.0
  * Assembly3 0.12.0
  * fasteners 0.4.75
  * Behave-Dark-Colors 0.1.1
  * freecad.gears 1.2.0
  * DynamicData 2.60.0
[/code]
```

### Subproject(s) affected?

PartDesign

### Anything else?

FreeCAD says, "Oh so you're trying to save time by moving more than 1 thing at a time? Oh Ok then, I'm gonna punish you for that by taking more time proportional to the items you tried to move, plus a bonus time penalty for good measure"

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2024-02-19:
@toddmo when the weekly builds today, can you please confirm that it fixes your issue? thanks!

closed 2024-02-19 by chennes

toddmo, 2024-02-19:
Yes Sir, I will. thank you very much and especially for getting it so quickly.

Thanks, Todd

commit 1c8ca27f (2024-02-19): Preserve object order when moving them

Remove and re-insert the selected objects at the specified point in the
tree while preserving their relative order.

The code used to insert them in reverse order.

Closes #12287