FreeCAD/FreeCAD #12288: [Problem] Sketcher - Disappearing element handles

royw, 2024-02-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When editing a sketch, the element handles disappeared.  This did not affect other sketches in the project.  The problem persists when FreeCAD is restarted.  Opened a forum thread, [Sketcher: Missing handles](https://forum.freecad.org/viewtopic.php?p=739129#p739129), where chrisb had seen this previously and suggested the work-around of setting the body visible, which worked.  This might be a regression. 

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35735 (Git) AppImage
Build type: Release
Branch: main
Hash: c12a5c2a163c1a277cf273106899209792be152d
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * CurvedShapes 1.0.4
  * Assembly4 0.12.5
  * A2plus 0.4.60h
  * fasteners 0.4.58
  * lattice2 1.0.0
  * Curves 0.6.12
```
```

### Subproject(s) affected?

None

### Anything else?

Forum thread: [Sketcher: Missing handles](https://forum.freecad.org/viewtopic.php?p=739129#p739129)
Test case: [Bug.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/14204152/Bug.FCStd.zip)
Image of issue: ![bug](https://github.com/FreeCAD/FreeCAD/assets/40304/52f2c3fc-83f6-4173-ade0-a74e60aac638)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-02-08:
This is a regression in the 0.22dev cycle:

My last Good build:

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35081 (Git) AppImage
Build type: Release
Branch: main
Hash: 32db044caa92a2adc448850179c524b793a03c3e
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * freecad.gears 1.2.0
```
My first Bad build:

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35274 (Git) AppImage
Build type: Release
Branch: main
Hash: dc063eabec61bffc72c92595c66a4649f25fe381
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * freecad.gears 1.2.0
```

Syres916, 2024-02-08:
Git Bisect shows root cause https://github.com/FreeCAD/FreeCAD/commit/a740632616cccc713281718fcee17501a1367202
I'll have a go at a fix and if successful submit a PR.

Syres916, 2024-02-08:
@Rexbas could you please have a look at this, in my simple mind, _src/Gui/NavigationStyle.cpp_ around line 573 needs changing from:

```
    // Fix issue with near clipping in orthogonal view
    if (camera->getTypeId().isDerivedFrom(SoOrthographicCamera::getClassTypeId())) {

        // The center of the bounding sphere in camera coordinate system
        SbVec3f center;

```

to:

```
    // Fix issue with near clipping in orthogonal view
    if (camera->getTypeId().isDerivedFrom(SoOrthographicCamera::getClassTypeId())) {

        if (isSketchInEdit(getActiveGuiDocument())) {
            SoGetBoundingBoxAction action(viewer->getSoRenderManager()->getViewportRegion());
            action.apply(viewer->getSceneGraph());
            SbSphere boundingSphere;
            boundingSphere.circumscribe(action.getBoundingBox());
        }

        // The center of the bounding sphere in camera coordinate system
        SbVec3f center;
```

But I can't seem to find the correct headers to include in order for that to work but you may well have a better method to achieve the correct result.

Rexbas, 2024-02-08:
Yes I will have a look

chennes, 2024-02-19:
@royw can you confirm that this is now fixed for you in the weekly generated today?

PR #12306 (2024-02-19): Gui: Fix disappearing element handles in sketcher
Finds the bounding sphere when the editing view provider is set. This ensures the right bounding sphere is used for finding the near and far clipping plane

Fixes #12288
closed 2024-02-19 by chennes