FreeCAD/FreeCAD #12300: [Problem] Hard Crash while idle and in background, v0.21.2-33771 on MacOS

iAmVaeVictus, 2024-02-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While the program is idle, and in the background (possibly while my machine was even with screen saver showing - not sure), the app crashes, and creates an Apple Crash Report (attached zip). 

File being edited is attached, zip. 

I wish I could give more steps, but that is it. Has repeated a couple of times in the past few hours. 

I think it was in the Part Designer workbench. 

### Full version info

```shell
[code]
OS: macOS 13.6.4
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

None

### Anything else?

[Switch Housing FOR FREECAD.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/14212379/Switch.Housing.FOR.FREECAD.FCStd.zip)
[CrashReportFreeCAD.txt.zip](https://github.com/FreeCAD/FreeCAD/files/14212437/CrashReportFreeCAD.txt.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-02-09:
> I think it was in the Part Designer workbench.

To me looks like _Sketcher_. 🤖  _CrashReportFreeCAD.txt_ contains: 
```cpp
Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
0   ???                                          0x1362e7d18 typeinfo for boost::uuids::basic_random_generator<boost::random::mersenne_twister_engine<unsigned int, 32ul, 624ul, 397ul, 31ul, 2567483615u, 11ul, 4294967295u, 7ul, 2636928640u, 15ul, 4022730752u, 18ul, 1812433253u>>::impl + 0
1   SketcherGui.so                               0x13b08976e SketcherGui::EditModeGeometryCoinManager::updateGeometryColor(Sketcher::GeoListModel<std::__1::unique_ptr<Sketcher::GeometryFacade const, std::__1::default_delete<Sketcher::GeometryFacade const>>> const&, bool) + 1934
2   SketcherGui.so                               0x13b085069 SketcherGui::EditModeCoinManager::updateColor(Sketcher::GeoListModel<std::__1::unique_ptr<Sketcher::GeometryFacade const, std::__1::default_delete<Sketcher::GeometryFacade const>>> const&) + 57
3   SketcherGui.so                               0x13b084fe6 SketcherGui::EditModeCoinManager::updateColor() + 54
4   SketcherGui.so                               0x13b0b2427 SketcherGui::ViewProviderSketch::onSelectionChanged(Gui::SelectionChanges const&) + 1319
5   libFreeCADGui.dylib                          0x10354fe28 Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&) + 40
6   libFreeCADGui.dylib                          0x1032a515e boost::signals2::detail::slot_call_iterator_t<boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type, Gui::SelectionChanges const&>, std::__1::__list_iterator<boost::shared_ptr<boost::signals2::detail::connection_body<std::__1::pair<boost::signals2::detail::slot_meta_group, boost::optional<int>>, boost::signals2::slot<void (Gui::SelectionChanges const&), boost::function<void (Gui::SelectionChanges const&)>>, boost::signals2::mutex>>, void*>, boost::signals2::detail::connection_body<std::__1::pair<boost::signals2::detail::slot_meta_group, boost::optional<int>>, boost::signals2::slot<void (Gui::SelectionChanges const&), boost::function<void (Gui::SelectionChanges const&)>>, boost::signals2::mutex>>::dereference() const + 78
7   libFreeCADGui.dylib                          0x1032a4c71 boost::signals2::detail::signal_impl<void (Gui::SelectionChanges const&), boost::signals2::optional_last_value<void>, int, std::__1::less<int>, boost::function<void (Gui::SelectionChanges const&)>, boost::function<void (boost::signals2::connection const&, Gui::SelectionChanges const&)>, boost::signals2::mutex>::operator()(Gui::SelectionChanges const&) + 673
8   libFreeCADGui.dylib                          0x103553d46 Gui::SelectionSingleton::slotSelectionChanged(Gui::SelectionChanges const&) + 550
9   libFreeCADGui.dylib                          0x1032a515e boost::signals2::detail::slot_call_iterator_t<boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type, Gui::SelectionChanges const&>, std::__1::__list_iterator<boost::shared_ptr<boost::signals2::detail::connection_body<std::__1::pair<boost::signals2::detail::slot_meta_group, boost::optional<int>>, boost::signals2::slot<void (Gui::SelectionChanges const&), boost::function<void (Gui::SelectionChanges const&)>>, boost::signals2::mutex>>, void*>, boost::signals2::detail::connection_body<std::__1::pair<boost::signals2::detail::slot_meta_group, boost::optional<int>>, boost::signals2::slot<void (Gui::SelectionChanges const&), boost::function<void (Gui::SelectionChanges const&)>>, boost::signals2::mutex>>::dereference() const + 78
10  libFreeCADGui.dylib                          0x1032a4c71 boost::signals2::detail::signal_impl<void (Gui::SelectionChanges const&), boost::signals2::optional_last_value<void>, int, std::__1::less<int>, boost::function<void (Gui::SelectionChanges const&)>, boost::function<void (boost::signals2::connection const&, Gui::SelectionChanges const&)>, boost::signals2::mutex>::operator()(Gui::SelectionChanges const&) + 673
11  libFreeCADGui.dylib                          0x1035523da Gui::SelectionSingleton::notify(Gui::SelectionChanges&&) + 362
12  libFreeCADGui.dylib                          0x103555b46 Gui::SelectionSingleton::rmvPreselect(bool) + 1286
13  SketcherGui.so                               0x13b05775e SketcherGui::TaskSketcherElements::onListWidgetElementsMouseMoveOnItem(QListWidgetItem*) + 126
14  libQt5Core.5.15.8.dylib                      0x105355f43 0x105209000 + 1363779
15  SketcherGui.so                               0x13b041efc SketcherGui::ElementView::onIndexHovered(QModelIndex) + 76
16  SketcherGui.so                               0x13b05dd6d QtPrivate::QSlotObject<void (SketcherGui::ElementView::*)(QModelIndex), QtPrivate::List<QModelIndex>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) + 125
17  libQt5Core.5.15.8.dylib                      0x105355f43 0x105209000 + 1363779
18  SketcherGui.so                               0x13b046d5c SketcherGui::ElementItemDelegate::editorEvent(QEvent*, QAbstractItemModel*, QStyleOptionViewItem const&, QModelIndex const&) + 1212
19  libQt5Widgets.5.15.8.dylib                   0x104a84f9a 0x1048f2000 + 1650586
```

The backtrace provided is really close to the one reported in #9198.

luzpaz, 2025-06-12:
@iAmVaeVictus can you still reproduce this in 1.0.1 ?