FreeCAD/FreeCAD #12308: [Problem] Path generates G-Code with 7+km radius arc

IdahoEv, 2024-02-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This appears to be an edge case where the path generator creates a large-radius arc (radius 7278224.337mm = 7.3km) rather than a short straight line segment.  While this will render ok in software simulations like Camotics, on at least some hardware (e.g. my Shapeoko 3) its endpoint is not handled correctly and causes the machine to travel in a straight line infinitely, running out-of-bounds.   

Here's the problematic gcode, for what is effectively a 3.4mm straight segment of a pocket.

```
G1 X121.919 Y-12.700 Z-3.175 F508.000 
G2 X121.919 Y-16.113 Z-3.175 I-7278224.337 J0.000 F1524.000 
```

### Reproduction steps 

1. Load the attached file in FC 0.21.2.  You may need A2Plus installed.
2. Switch to Path workbench
3. Select the Job in the combo view
4. Click "Post Process"
5. Save the .nc file in a convenient location
6. Open the .nc file in a text editor and look at line 25 for a large I parameter

### Further comments

I think the first step is testing this against 0.21.1 to look for regressions, because I see that code related to detecting arcs, lines, and tolerances in Path workbench was upgraded and refactored in early January 2024.   I want to test that myself but so far I have not been able to build FreeCAD from source to have a second version for testing.    I will look for another machine I can test it on and will post a comment or update if I manage that.  

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.62
  * Assembly3 0.12.2 (Disabled)
  * Assembly4 0.50.6
  * fasteners 0.4.63
  * FeedsAndSpeeds 0.5.0 (Disabled)
  * Manipulator 1.5.7 (Disabled)
[/code]
```

### Subproject(s) affected?

Path

### Anything else?

[Forum discussion](https://forum.freecad.org/viewtopic.php?t=84888)
[Arc radius bug case.zip](https://github.com/FreeCAD/FreeCAD/files/14215773/Arc.radius.bug.case.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

IdahoEv, 2024-02-09:
I have now also reproduced this in 0.21.1.  After seeing some commits in possibly-related code between 0.21.1 and 0.21.2 I wanted to see if it was a regression between versions.  I uninstalled 0.21.2 and reinstalled 0.21.1 from a fresh download.  The bad g-code still appears with this file.   

[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.62
  * Assembly3 0.12.2 (Disabled)
  * Assembly4 0.50.6
  * fasteners 0.4.63
  * FeedsAndSpeeds 0.5.0 (Disabled)
  * Manipulator 1.5.7 (Disabled)
[/code]

luzpaz, 2024-03-12:
@IdahoEv Can you reproduce on development version ? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

maxwxyz, 2024-11-01:
@IdahoEv can you try in the latest RC candidates maybe?

kalbfuss, 2025-04-26:
I have tested with the latest FreeCAD Daily built and confirm that the issue still exists. Here is the "About" from the version I have used: 

```
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40971 (Git)
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: German/Switzerland (de_CH)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/141.951/1
Installed mods: 
  * Curves 0.6.51
```

I ran into the issue when creating paths for the following model:

[Linke Seite.zip](https://github.com/user-attachments/files/19922017/Linke.Seite.zip)

FreeCAD produces the following arc command in line 269 of the resulting G-code file, which cannot be handled by my GRBL controller:

```
G2 X105.784 Y121.395 Z-1.000 I3251415.739 J1599160.045 K0.000 F1000.000
```

The result is a linear, infinite movement into a certain direction until the limit switches trigger. Commenting out the line or reducing the coordinate values resolves the problem.

I am using the default controller, which ships with the Genmitsu 4040 Pro Max CNC router. This is the version information returned by the controller:

```
Grbl 1.1f ['$' for help]
Grbl for ARM32
Version:ARM32 V3.4

> $I
[VER:ARM32 V3.4.20240825:]
[OPT:VZL,35,254]
ok
```

I have not looked into the GRBL source code, but I suspect the controller uses fixed point numbers. Assuming 3 decimal paces, the maximum supported coordinate value should be in the order of +/- 2^21 = +/- 2097152 for a 32 bit controller. This value is clearly exceeded.

Since the movement for very large coordinate values anyway becomes linear, a simple fix could be to replace the arc movement by a linear movement if a certain threshold is exceeded.

luzpaz, 2025-05-05:
@LarryWoestman any chance you can weigh-in ?

LarryWoestman, 2025-05-05:
As far as I can tell, the issue is with the Path generation code in the CAM workbench rather than in the postprocessor code.  The postprocessor is just using the numeric values it is given.  While it might be possible to do some range checking in the postprocessor, the real issue needs to be fixed in the Path generation code (which I know very little about).

michaelosthege, 2025-06-07:
Still happening in v1.0.0.

Here's a minimal example reproducing the bug: 
[FreeCAD_issue_12308.zip](https://github.com/user-attachments/files/20638447/FreeCAD_issue_12308.zip)

My GRBL also got scared and did the infinite linear movement 👇 fortunately I was just testing my speeds..

[image]

I think it happens when a ZigZagOffset part perfectly hits the corner:

[image]

michaelosthege, 2025-06-07:
@tarman3 correction: I'm already on v1.0.1

michaelosthege, 2025-06-08:
The bug is still there in the weekly build

[image]

luzpaz, 2025-06-08:
@michaelosthege can you add your full [About](https://wiki.freecad.org/About) info please ? TIA

tarman3, 2025-06-08:
Can not reproduce
Did you try to recompute?

tarman3, 2025-06-08:
Path after recompute
[FreeCAD_issue_12308.txt](https://github.com/user-attachments/files/20645344/FreeCAD_issue_12308.txt)

<details>
  <summary>FreeCAD 1.0.1.39285 x86_64 ArchLinux KDE wayland</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.1.39285 (Git)
Build type: Release
Branch: makepkg
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: SearchBar 1.4.2 (Disabled); CurvedShapes 1.0.13; lattice2 1.0.0; Silk 0.1.9; freecad.gears 1.3.0; MeshRemodel 1.10.37; Curves 0.6.63; sheetmetal 0.7.22; SaveAndRestore 0.2.1 (Disabled); Defeaturing 1.2.6

```

</details>

tarman3, 2025-06-09:
There is another bug
Workaround is enable `SplitArc` 

[image]

tarman3, 2025-06-09:
I dont know
Path was generate in C++ files
For me this is a incomprehensible dark box

sliptonic, 2025-10-16:
I can't duplicate the problem with the supplied file.  
I suspect the problem is arc tolerance as noted in this forum post:
https://forum.freecad.org/viewtopic.php?t=100426&sid=2a4224427f9dd7757d7f0027aed531bd&start=10#p851911

Closing this issue.  Please comment if that isn't the source of the issue and upload a new test file.