FreeCAD/FreeCAD #12316: [Problem] Adjacent edge selection

X-Ryl669, 2024-02-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When selecting edges for filleting or chamfering, it's a tedious "1px hit process". It's even harder with HiDpi screen.

I was wondering if it was possible to add a feature / button or shortcut to select adjacent edge to the last selected one that lies on the same plane. Typically, when you're selecting the edges of some face, you will follow selecting the next sibling edge and you'll usually keep selecting the edges on the same plane. The process will stop when it reaches the first selected edge (full connectivity).

Typically the process will be:
1. You select the first edge [image]

2. You click the button/press the shortcut
3. The algorithm will select the next edge, like this: 
[image]

4.  And so on until all edges are selected: 
[image]

5. The example I've chosen is ambiguous, since the algorithm could have selected this edge instead to continue: 
[image]
The ambiguity can be removed if we started by the round edge (since any sampling of 3 points the edge gives a plane that we can use as a selection filter for edges), or by selecting edges manually until there's an non zero angle between their directions.

Another possibility is to select one edge, then the face that's normal to the edge and that will select all connected edges that lies on that face and touch the previous selection, iteratively until the next edge is the one you started selecting from.
A bit like this:
1. You select the first edge [image]
2. You press the shortcut or the button and then select the face you want the selection to happen on: 
[image]
4. The algorithm is then able to select all edge successively that touch the previous selected edge and lies on the selected plane
[image]
 

It would be a huge time saver and UX improvement.

### Full version info

```shell
0.21.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-02-09:
Loop selection has already been added to the 0.22dev builds https://github.com/FreeCAD/FreeCAD/pull/11925 , it won't get backported to 0.21.x because it's not a fix, it's a new feature. You can try the feature in one of the weekly builds https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds but remember there's lots of changes going on so regressions are still around, the warning on the splash screen should be heeded.

X-Ryl669, 2024-02-09:
As far as I understand your link, this is for the sketcher, not for the normal object. I've just tried the weekly build and it doesn't work. If you double click on any edge in Part Design's body (as in the first picture above) it selects the whole body and not the connected edges for the rounded rectangle.

Roy-043, 2024-02-09:
For operations such as filleting selecting one edge is enough if edges are tangential as in the example.

There already is this:
https://wiki.freecad.org/Path_SelectLoop

X-Ryl669, 2024-02-09:
Wonderful! Is it possible to trigger this in the part design workbench ? Right now, I've tested it and it works, but I need to change to the Path workbench first.

Roy-043, 2024-02-10:
You can create a custom toolbar with this tool. You may then also need to auto-load the Path workbench at startup.
https://wiki.freecad.org/Interface_Customization#Toolbars
https://wiki.freecad.org/Preferences_Editor#Workbenches

X-Ryl669, 2024-02-12:
Is there anything specific to Path for this tool? Is is possible to move it to a common tool (it's useful in many workbench, Part Design, Part, Path, ...) ?

Selecting a closed edge part by part is a real PITA from a UX perspective, you have to move the camera, select, try to be 1px on the edge, if there is some occlusion, you're doomed. I don't see that in other CAD software.

maxwxyz, 2024-03-02:
Related / would benefit this problem: https://github.com/FreeCAD/FreeCAD/issues/10308

maxwxyz, 2024-08-30:
@pierreporte would this be CWG related? Would be very helpful to select quickly closed profiles on a sketch for extrusion

pierreporte, 2024-08-30:
@maxwxyz For me it’s UX. Definitely helpful though.

X-Ryl669, 2025-04-08:
@maxwxyz : Can you add this to the projects [UI / UX Improvements](https://github.com/orgs/FreeCAD/projects/24) list, please ?

karamellpelle, 2026-01-06:
My explaination above of the [edge loop algorithm](https://docs.blender.org/manual/en/latest/modeling/meshes/selecting/loops.html) may be over-detailed. The algorithm is pretty easy: If the number of edge choices at a vertex is 0, 1, 2 or 3, then pick the edge with direction that gives the highest dot product compared to current direction, and continue. Otherwise stop. Also make sure to stop if we get back to a previous vertex.

This would probably be easy to implement.

Using a modifier like <kbd>Alt</kbd> when clicking edges will be better than a button. And using <kbd>Shift</kbd>+<kbd>Alt</kbd> to add the loop to current selection will efficiently solve ambiguity at vertexes.

X-Ryl669, 2026-01-07:
> I didn't understand the problem with the edge along extrusion.

There's no reason the edge loop you want to follow is the one on the top face. One could want to select the edge loop on the vertical face (imagine the cube isn't cut with a round shape but a plain rectangle and you need to chamfer the internal cut).

All in all, selecting edge loop isn't possible with a single edge selection, whatever you the algorithm you come up to. You need to account for the "main" face the edges are on (either by selecting 2 non colinear edges, or selecting a face, or selecting 2 faces to get the edges they are intersecting with, ...). So the edge selection algorithm is a 2D (3D) algorithm, not a 1D curve algorithm like you are proposing.

Again, the algorithm for edge selection works and it's already in FreeCAD since 0.22 (no need for macro or anything else). The issue is that it's un-discoverable and so, unusable. That's a bad user experience and that need to be fixed by default, not by expecting users to auto discover the trick and make a "custom button palette" with the feature, or long clicking/right clicking the edge. Shift clicking and Ctrl clicking is already used in the interface, so maybe Alt clicking should be reserved for this (or Ctrl + Shift clicking).