FreeCAD/FreeCAD #12318: [Problem] Freecad doesn't detect Windows dark mode.

MisterMakerNL, 2024-02-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

So I was looking into making default stylesheets.
Which was nothing more than the no-stylesheet but then in dark. And with some bug fixes.
And I came across a little [python QT mod](https://github.com/5yutan5/PyQtDarkTheme/tree/main).
This is using a python [module called darkdetect](https://pypi.org/project/darkdetect/).
Also @sliptonic is pushing dark mode as default with Ondsel.
Adding a dark detect for windows would fix the issue of getting into fights about what should be default.
I can make the dark windows stylesheet.
I wanted to test python module myself but got errors on compiling that I was missing the module. But on installation it said it was already installed. So ugh.
Hopefully it works.

### Full version info

```shell
NA
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-02-09:
Related to https://github.com/FreeCAD/FreeCAD/issues/11984

luzpaz, 2024-08-27:
@obelisk79 any idea what's the status of this ?

MisterMakerNL, 2024-08-27:
There is btw a dark style detection in overlay:
https://github.com/FreeCAD/FreeCAD/blob/73cb5bafe0c32c178d9b594a6ca03124483d305f/src/Gui/OverlayManager.cpp#L141

luzpaz, 2024-08-30:
@kadet1090 can you weigh-in ?

closed 2025-08-26 by github-actions[bot]