FreeCAD/FreeCAD #12322: [Problem] V0.22 Sketcher OVP: sign ignored for width and height input for line

Roy-043, 2024-02-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Set the On-View-Parameters preference to "All".
2. Create a new sketch and start Sketcher_CreateLine.
4. Select Mode "Point, width, height".
5. For the start point pick the origin (in practice this might be another already constrained point).
6. Move the cursor to the left of the Y axis and below the X axis.
7. The distances are now displayed with a minus sign.
8. Enter negative values for the X and Y distance.
9. Result: The endpoint has positive coords.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35803 (Git)
Build type: Release
Branch: main
Hash: 90c90f705bec20faf4b8bb419574134c33c7d4b5
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-02-09:
@Roy-043 are you seeing the same behaviour for real geometry and construction, the reason I ask is the investigation I've started into https://forum.freecad.org/viewtopic.php?t=85137 ?

Roy-043, 2024-02-09:
@Syres916: Same behavior for contruction geometry.

Syres916, 2024-02-09:
Well, I'll add your test to each step of my Git Bisect test for the regression that I noted and we'll see if it's all down to the same commit.

Syres916, 2024-02-10:
This bug isn't triggered by the same commit that the construction geometry display bug is but it's very close, most likely the same day. So I'll carry on going through the 7 commits later this afternoon.

PaddleStroke, 2024-02-12:
Regarding the issue, this was the original behavior of rectangle as well. But people asked for it to be changed. But line tool has indeed not be changed.
I will do the change.

closed 2024-02-28 by chennes