FreeCAD/FreeCAD #12323: [Problem] V0.22 Sketcher OVP: inputting start angle for arc does not result in constraint

Roy-043, 2024-02-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Set the On-View-Parameters preference to "All".
2. Create a new sketch and start Sketcher_CreateArc.
4. Enter all OVP values.
5. Result: for all entered values, except the start angle, constraints are created. This is inconsistent. Compare 3 points rectangle where the start angle does result in a constraint.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35803 (Git)
Build type: Release
Branch: main
Hash: 90c90f705bec20faf4b8bb419574134c33c7d4b5
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-06-28:
@PaddleStroke

FEA-eng, 2025-02-11:
Still relevant