FreeCAD/FreeCAD #12326: Part Container breaks Transformations

HansCarstorp, 2024-02-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Cleaning up your document by moving constructions of part-design into a part container, especially moving sketches and shape binders in and out breaks the document and freecad starts to crash when trying to fix the part design bodies.

Form the observed behavior I would guess that this is a deep and hard to fix problem. Transformations in presence of containers that can be nested like a "Parts" do change the order of transformations when moved in and out of parent containers to clean things up. 

The problem fix would be to have transformations removed as properties of objects and become features like mirror or the rotation pattern in the part- design workbench. In that case a the transform order and rotation order is clearly given by the nesting hierarchy of the Part containers.  Having freecad using position and rotation as properties of the object is, even if it would not created broken objects going to confuse when cleaning up documents. 

I am sure that is a duplicate of some reported bug, but I wanted to suggest the above fix.
In any case a transform should be visible in the document like other features or modifications and not be an attribute of anything.  This could be a feature request as well, but I already lost so much time with freecads issues. 

### Full version info

```shell
All versions I tested including derivations like freecad daily. This has not been fixed for years.
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-02-10:
your issue is hard to understand. You may be attempting to do things that are not supported such as dragging features out of Bodies. To clarify please provide a file and step-by-step instructions.

maxwxyz, 2024-02-20:
@HansCarstorp please provide more info

luzpaz, 2024-03-09:
Closing as no response from OP. Please re-open if there is a follow-up

closed 2024-03-09 by luzpaz